CCNYRoboticsLab/imu_tools #40: Orientation Covariance

paulbovbel, 2015-01-27:
Currently no covariance is provided for the output orientations. Short of a proper covariance propagation from the original sensor sources, I'd like to stub in a parametrized reasonable fixed covariance matrix, so that the output can be used with robot_localization and robot_pose_ekf

mintar, 2015-01-28:
At least for robot_pose_ekf, that's not necessary. If the (0, 0) element of the covariance matrix is 0, it uses a default one of 0.01 deg/s: [odom_estimation_node.cpp:253](https://github.com/ros-planning/navigation/blob/9b92c2c08392f05c71bce1175a6e4ea343248db8/robot_pose_ekf/src/odom_estimation_node.cpp#L253). It seems better to me to leave the matrix all zeroes than to make one up, so the subscribers know that the covariance is unknown. From Imu.msg:

> A covariance matrix of all zeros will be interpreted as "covariance unknown", and to use the
> data a covariance will have to be assumed or gotten from some other source.

mintar, 2015-01-28:
BTW, we've been successfully using imu_filter_madgwick in combination with robot_pose_ekf for years.

mintar, 2015-01-28:
On second thought, this *does* sound like a useful feature, since it allows control over the covariances that go into robot_pose_ekf. What we could do:

* add a dynamic reconfigure param, say `orientation_covar_stddev`
* put `orientation_covar_stddev^2` on the diagonal of the orientation_covariance matrix

If we leave the default value of orientation_covar_stddev set to 0, this keeps the current behavior, while allowing the user to tune the covariances until they get good results out of robot_pose_ekf. What do you think?

paulbovbel, 2015-01-28:
Sounds like a plan

PR #41 (2015-01-29): Covariance param
Fix #40
closed 2015-01-29 by mintar