CCNYRoboticsLab/imu_tools #45: Provide C++ API

mistoll, 2015-07-24:
It would be usefull to have a C++ API to the madgwick algorithm. 

Should be fairly easy to implement by splitting ImuFilter class. If you like, I'll submit a pull request.

mintar, 2015-07-24:
Sure, go ahead! I'm on vacation for the next 2 weeks, so expect delays in answering. :-)

mistoll, 2015-07-29:
Ok, here's a first impression. 
But I don't like the interface, because there are to many parameters in the public members. Tf2 oder message types would be handy there. What do you think?
Another function I'd like to provide a public API to is computeRPY. Despite this is rather simple it is still a imu to orientation filter. So this could go into another class (like ImuFilterStateless).
Then you could even provide a parameter to switch between different filter implementations - but that's beyond what I need.

BTW have a nice vacation ;-)

mintar, 2015-08-25:
Ok, I finally got a chance to review #46 . It looks very good to me, ready to merge.

I do like the interface and wouldn't use ROS types to wrap the parameters. The nice thing about the interface as proposed in #46 is that the new ImuFilter class doesn't include anything from ROS, so in the future we could even move it into a standalone library so that it can be reused from other robotics frameworks.

About computeRPY: It is pretty much a hack that's been tacked onto the original Madgwick implementation. In the original implementation, the orientation was initialized to all zeros, which caused the filter to slowly converge (over the course of about 10 seconds) to the correct orientation; computeRPY fixes that.

However, computeRPY is buggy: sometimes it computes an incorrect orientation. Since computeRPY is only used for initialization, the filter will still converge to the correct orientation in the end, and having it is better than nothing. However, I think it should be completely rewritten to use quaternions instead of RPY, and to use the same math as the rest of the filter. I tried once, but failed. For these reasons, I wouldn't want to provide a public interface to computeRPY at the moment.

From my point of view, #46 is ready to merge. Do you still want to add something?

mistoll, 2015-08-25:
Sure, if you like it, go ahead, merge it.

mintar, 2015-08-25:
Done! Thanks again!

closed 2015-08-25 by mintar