CCNYRoboticsLab/imu_tools #47: Problem installing imu_madgwick_filter on raspberry PI

MauroPagliarella, 2015-09-19:
Hi all
I'm building a robot running ROS on board using a Raspberry PI. 

SYSTEM CONFIGURATION:
Raspberry PI running ubuntu wheezy and custom ROS Indigo.

PROBLEM:
I tried to install the imu_filter_madgwick package but failed. It seems to be necessary to install rviz as a prerequisite for the installation of the imu_filter_madgwick. Since I have a custom installation, I followed the procedure described in 

http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Indigo%20on%20Raspberry%20Pi 

To be precise, I used the following commands to download the packages and install it:
1: cd ROS_Comm
2: rosinstall_generator ros_comm rviz imu_tools --rosdistro indigo --deps --wet-only --exclude roslisp --tar > indigo-custom_ros.rosinstall
3: wstool merge -t src indigo-custom_ros.rosinstall
4: wstool update -t src
5: rosdep install --from-paths src --ignore-src --rosdistro indigo -y -r --os=debian:wheezy
6: sudo apt-get update
7: sudo ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/indigo

The error log is reported below.

QUESTION:
Is it a way to install only imu_madgwick_filter package without install imu_tools and rviz packages?

Any help is appreciated. Thank you very much.
Mauro

////@@@@
Error log:
==> make -j4 -l4 in '/home/pi/ROS_Comm/build_isolated/rviz'
Scanning dependencies of target interactive_marker_test
[  0%] Built target connect_test
[  0%] Building CXX object src/test/CMakeFiles/interactive_marker_test.dir/interactive_marker_test.cpp.o
Scanning dependencies of target rviz
[  0%] [  0%] [  0%] Building CXX object src/rviz/CMakeFiles/rviz.dir/frame_manager.cpp.o
Building CXX object src/rviz/CMakeFiles/rviz.dir/frame_position_tracking_view_controller.cpp.o
Building CXX object src/rviz/CMakeFiles/rviz.dir/image/ros_image_texture.cpp.o
Linking CXX executable /home/pi/ROS_Comm/devel_isolated/rviz/lib/rviz/interactive_marker_test
/tmp/ccX1hvNB.s: Assembler messages:
/tmp/ccX1hvNB.s:2010: Warning: swp{b} use is deprecated for ARMv6 and ARMv7

...

/tmp/ccX1hvNB.s:7021: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccX1hvNB.s:7038: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
[  0%] Built target interactive_marker_test
[  0%] [  0%] Building CXX object src/rviz/CMakeFiles/rviz.dir/image/image_display_base.cpp.o
Building CXX object src/rviz/CMakeFiles/rviz.dir/properties/tf_frame_property.cpp.o
/tmp/ccJX50cx.s: Assembler messages:
/tmp/ccJX50cx.s:1336: Warning: swp{b} use is deprecated for ARMv6 and ARMv7

...

/tmp/cc2QOWZ4.s:899: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
Linking CXX shared library /home/pi/ROS_Comm/devel_isolated/rviz/lib/librviz.so
[ 55%] Built target rviz
Scanning dependencies of target rviz_image_view
Linking CXX executable /home/pi/ROS_Comm/devel_isolated/rviz/lib/rviz/rviz
Linking CXX executable /home/pi/ROS_Comm/devel_isolated/rviz/lib/rviz/color_editor_test
[ 55%] Building CXX object src/image_view/CMakeFiles/rviz_image_view.dir/image_view.cpp.o
Scanning dependencies of target default_plugin
/home/pi/ROS_Comm/devel_isolated/rviz/lib/librviz.so: undefined reference to `vtable for Assimp::IOSystem'
/home/pi/ROS_Comm/devel_isolated/rviz/lib/librviz.so: undefined reference to `typeinfo for Assimp::IOSystem'
collect2: ld returned 1 exit status
src/rviz/CMakeFiles/executable.dir/build.make:212: recipe for target '/home/pi/ROS_Comm/devel_isolated/rviz/lib/rviz/rviz' failed
make[2]: *** [/home/pi/ROS_Comm/devel_isolated/rviz/lib/rviz/rviz] Error 1
CMakeFiles/Makefile2:1659: recipe for target 'src/rviz/CMakeFiles/executable.dir/all' failed
make[1]: *** [src/rviz/CMakeFiles/executable.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 56%] /home/pi/ROS_Comm/devel_isolated/rviz/lib/librviz.so: undefined reference to `vtable for Assimp::IOSystem'
/home/pi/ROS_Comm/devel_isolated/rviz/lib/librviz.so: undefined reference to `typeinfo for Assimp::IOSystem'
collect2: ld returned 1 exit status
src/test/CMakeFiles/color_editor_test.dir/build.make:214: recipe for target '/home/pi/ROS_Comm/devel_isolated/rviz/lib/rviz/color_editor_test' failed
make[2]: *** [/home/pi/ROS_Comm/devel_isolated/rviz/lib/rviz/color_editor_test] Error 1
CMakeFiles/Makefile2:2148: recipe for target 'src/test/CMakeFiles/color_editor_test.dir/all' failed
make[1]: *** [src/test/CMakeFiles/color_editor_test.dir/all] Error 2
Building CXX object src/image_view/CMakeFiles/rviz_image_view.dir/main.cpp.o
[ 56%] Building CXX object src/image_view/CMakeFiles/rviz_image_view.dir/moc_image_view.cxx.o
[ 56%] Building CXX object src/rviz/default_plugin/CMakeFiles/default_plugin.dir/axes_display.cpp.o
/tmp/ccDnygzd.s: Assembler messages:
/tmp/ccDnygzd.s:1063: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccDnygzd.s:1073: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccDnygzd.s:1090: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
Linking CXX executable /home/pi/ROS_Comm/devel_isolated/rviz/lib/rviz/image_view
/tmp/ccMtlwCs.s: Assembler messages:
/tmp/ccMtlwCs.s:555: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccMtlwCs.s:565: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccMtlwCs.s:582: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
[ 57%] Building CXX object src/rviz/default_plugin/CMakeFiles/default_plugin.dir/depth_cloud_display.cpp.o
/home/pi/ROS_Comm/devel_isolated/rviz/lib/librviz.so: undefined reference to `vtable for Assimp::IOSystem'
/home/pi/ROS_Comm/devel_isolated/rviz/lib/librviz.so: undefined reference to `typeinfo for Assimp::IOSystem'
collect2: ld returned 1 exit status
src/image_view/CMakeFiles/rviz_image_view.dir/build.make:264: recipe for target '/home/pi/ROS_Comm/devel_isolated/rviz/lib/rviz/image_view' failed
make[2]: *** [/home/pi/ROS_Comm/devel_isolated/rviz/lib/rviz/image_view] Error 1
CMakeFiles/Makefile2:1794: recipe for target 'src/image_view/CMakeFiles/rviz_image_view.dir/all' failed
make[1]: *** [src/image_view/CMakeFiles/rviz_image_view.dir/all] Error 2
[ 57%] Building CXX object src/rviz/default_plugin/CMakeFiles/default_plugin.dir/effort_display.cpp.o
[ 57%] Building CXX object src/rviz/default_plugin/CMakeFiles/default_plugin.dir/depth_cloud_mld.cpp.o
[ 57%] Building CXX object src/rviz/default_plugin/CMakeFiles/default_plugin.dir/camera_display.cpp.o
/tmp/cctioS6c.s: Assembler messages:
/tmp/cctioS6c.s:693: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/cctioS6c.s:702: Warning: swp{b} use is deprecated for ARMv6 and ARMv7

...

/tmp/cctioS6c.s:7239: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/cctioS6c.s:7249: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
[ 57%] Building CXX object src/rviz/default_plugin/CMakeFiles/default_plugin.dir/effort_visual.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.6/README.Bugs> for instructions.
src/rviz/default_plugin/CMakeFiles/default_plugin.dir/build.make:356: recipe for target 'src/rviz/default_plugin/CMakeFiles/default_plugin.dir/effort_display.cpp.o' failed
make[2]: *** [src/rviz/default_plugin/CMakeFiles/default_plugin.dir/effort_display.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
/tmp/ccj6TSGG.s: Assembler messages:
/tmp/ccj6TSGG.s:1591: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccj6TSGG.s:1601: Warning: swp{b} use is deprecated for ARMv6 and ARMv7

...

/tmp/cczRZE7s.s:80706: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/cczRZE7s.s:80723: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
CMakeFiles/Makefile2:1744: recipe for target 'src/rviz/default_plugin/CMakeFiles/default_plugin.dir/all' failed
make[1]: *** [src/rviz/default_plugin/CMakeFiles/default_plugin.dir/all] Error 2
Makefile:123: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'rviz': 
  Command '['/opt/ros/indigo/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/pi/ROS_Comm/build_isolated/rviz && /opt/ros/indigo/env.sh make -j4 -l4

Command failed, exiting.
pi@raspberrypi ~/ROS_Comm $

mintar, 2015-09-21:
> QUESTION:
> Is it a way to install only imu_madgwick_filter package without install imu_tools and rviz packages?

Yes, you can directly specify a single package (`rosinstall_generator imu_madgwick_filter ...`) instead of the whole repo, which includes all 3 packages (`rosinstall_generator imu_tools ...`). That remove the dependency on rviz.

MauroPagliarella, 2015-09-21:
I already tried but it didn't work. It tried to install all packages anyway. I'll try again, maybe I did some error.
Thank you for your help. I'll keep you informed.

closed 2015-10-06 by mintar