CCNYRoboticsLab/imu_tools #48: [question] Void messages in /imu/data topic

carlosjoserg, 2015-09-24:
Hi, I'm trying the `imu_filter_madgwick` installed from synaptics with all default parameters (with out Mag.)

I publish the sensor data in `/imu/data_raw` and messages are there, I can see how the acceleration vector changes according to the orientation.

I see in the node graph that the `imu_filter_madgwick` is subscribed to that topic, so it should be receiving what I'm publishing (I also tried the nodelet interface)

I can see that the `/imu/data` topic is there with `rostopic list` but nothing is coming from there when I use `rostopic echo /imu/data`. In fact, that topic does not appear in the node graph.

Am I missing something in the setup? 
For instance, I see in Seb's videos that he starts with a certain configuration of the IMU and wait some seconds before moving it around.

carlosjoserg, 2015-09-24:
Forgot to say, Ubuntu 14.04/Indigo

mintar, 2015-09-24:
Could you record a rosbag of the data_raw topic please?
-- 
Diese Nachricht wurde von meinem Android-Mobiltelefon mit K-9 Mail gesendet.

carlosjoserg, 2015-09-24:
[Here](https://www.dropbox.com/s/47fefskc4a4w8oj/2015-09-24-23-58-09.bag?dl=0) it is.

I moved from the initial pose, twice to one direction, and twice to the other direction, and finish at the same initial position.

mintar, 2015-09-25:
With the following commands, I was able to run everything without problems:

    $ roscore
    $ rosparam set use_sim_time true
    $ rosbag play --clock --loop 2015-09-24-23-58-09.bag
    $ roslaunch imu.launch

This is the imu.launch I used:

    <launch>
      
      <node pkg="nodelet" type="nodelet" name="imu_manager" 
        args="manager" output="screen" />
    
      
      <node pkg="nodelet" type="nodelet" name="ImuFilterNodelet" 
        args="load imu_filter_madgwick/ImuFilterNodelet imu_manager" 
        output="screen">
    
        <param name="use_magnetic_field_msg" value="false"/>
        <param name="use_mag" value="false"/>
        <param name="fixed_frame" value="odom" />
        <param name="publish_tf" value="true" />
      </node>
    </launch>

The `fixed_frame` and `publish_tf` params are just for debug, so you can visualize the `tf` topic in RViz (you need to set the RViz fixed frame to `odom`). In a real system, you'd feed the `/data/imu` topic into something like `robot_pose_ekf` instead, which in turn would produce the proper tfs.

By the way, the messages in your bagfile have `frame_id` "world"; this has to be the IMU frame, so it's a bit of a strange name for that.

carlosjoserg, 2015-09-25:
Ok, I was launching similarly but without setting any parameter, i.e. using the default ones.

The only difference is that you explicitly set `use_mag` parameter to false, and it is default to true [here](https://github.com/ccny-ros-pkg/imu_tools/blob/indigo/imu_filter_madgwick/src/imu_filter_ros.cpp#L40). I don't see how that can affect, but now it is working here as well.

Thanks @mintar, I'll close the issue.

closed 2015-09-25 by carlosjoserg