CCNYRoboticsLab/imu_tools #53: How to tell if magnetometers are working

phyatt16, 2015-10-09:
I would like to use your filter for several IMUs to find the rotation from one IMU to another. This requires that they all have the same base or world frame - is that true for this filter?

I thought it should be true of this filter as long as I am using the magnetometer data, but how do I tell if I am using the magnetometer data or not? My magnetometer data is being published to the ros topic "/imu/mag" as a Vector3 Stamped type message. Can the code in its current state use this magnetometer information?

I have been playing with this for hours, but cannot get any of my IMUs to have the same orientations even when I can see them both sitting in front of me in the same orientation. I'm not sure what the problem is.

Thanks.

carlosjoserg, 2015-10-10:
Hi @phyatt16 

We are working on the relative IMU problem to sense a robotic hand configuration, but we haven't been able to use the magnetometer information anyway due to several metal pieces around, motor, etc. If that is your case too or similar, and you manage how to deal with that, we'd like to see it.

So, we've managed to do the relative IMU orientation with a slight modification of Madgwick's algorithm. As far as I understand from your approach, the difference is that you'd obtain the relative orientation as a post-process of the filter's output, whereas we'd directly provide the relative orientation as the filter's output, am I right?

If you have acecss to IEEExplore, [here](http://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=7139569) is the related paper. We hope to release the filter soon, and perhaps we might include it here if the ccny people likes to, but so far is private.

/cc @GaspareSantaera

robertogl, 2015-10-10:
Hi 
this new version of the filter subscrisbes to MagField message instead of a Vector3 bit if this is what your sensor's driver piblishes you can easily change the code by changing type of message.
However, the interferences of an indoor environment will not make your goal easy. I suggest you not to use magnetic field but in that case make sure all your devices start with the same exact orientation.
Please let ne know if that solves your problems or if you need any help.

Cheers,
Roberto

closed 2016-04-22 by mintar