CCNYRoboticsLab/imu_tools #54: released package imu_complementary_filter fails to build

dirk-thomas, 2015-10-15:
http://jenkins.ros.org/view/IbinT64/job/ros-indigo-imu-complementary-filter_binarydeb_trusty_amd64/

It seems like it does not declare a dependency on Eigen.

closed 2015-10-16 by mintar

dirk-thomas, 2015-10-16:
Thanks for the fast fix. Please watch out for the notifications emails sent by the build farm since we commonly don't create tickets for these things.

robertogl, 2015-10-16:
Thank you guys, sorry If I didn't fix it, I was planning to do it this
afternoon. Thanks Martin.

mintar, 2015-10-16:
@dirk-thomas Yes, thanks again for spotting this and opening the ticket. The problem with the notification emails is the high signal-to-noise ratio: At least 90% of the mails I receive are false positives that are caused by build farm failures (apt-get, ...), so it's hard to spot the true positives.

dirk-thomas, 2015-10-16:
For the specific job in question (http://jenkins.ros.org/view/IbinT64/job/ros-indigo-imu-complementary-filter_binarydeb_trusty_amd64/) it looks like that the daily build / email was always a real error.

Which job are you referring to with the high number of false positives?

mintar, 2015-10-17:
These are the two that are failing at the moment:

http://jenkins.ros.org/job/doc-indigo-rviz_fps_plugin/79/
http://jenkins.ros.org/job/doc-indigo-slam6d_exporter/18/

If you only look at the build history, most builds are passing. But I don't get emails for those, of course. The percentage of false positives is so high since most of our repos are quite stable and didn't get any commits lately, so we didn't break stuff, which only leaves the false positives.

Anyhow, I'll pay more attention to the mails in the future. And thanks again for opening this issue, it would have taken me much longer to notice otherwise.

dirk-thomas, 2015-10-18:
The error `Hash Sum mismatch` is coming from `apt-get` when the repository is changed while the cliient runs an update. The current build farm can't work around this. The new build farm (which is still not deployed) will automatically retry and not bother the developers with emails about it. These kind of errors happen mostly when a big rebuild is ongoing which commonly happens only every few months.

When ever you receive an email about a failing job you will also receive an email about the next job being successful again. So if the error seems to be unrelated to your code just wait for next build (depending on the job type: source 15 min., binary 1 day, doc 3 days) to decide if it fixed itself.

mintar, 2015-10-18:
Yes, that's what I do. Only sometimes I then forget about the mail and don't notice that the "back to normal" mail doesn't arrive. Three days is a long time to remember this stuff.

commit e4324d8f (2015-10-16): complementary: Add Eigen dependency

Fixes #54.