CCNYRoboticsLab/imu_tools #59: bug in rviz_imu_plugin.

asimay, 2016-06-08:
I first add rviz_imu_plugin, and then when I click remove button, the rviz crash... I tried many times, every time it is crash..

asimay, 2016-06-08:
and next potential bug: I add one rviz_imu_plugin for imu_topic, and then create another for imu_filtered_topic, the two cannot behave normally sometimes. sometimes the arrow of one can appear, the arrow of another cannot appear.

mintar, 2016-06-08:
Which ROS version? OS version? Which version of imu_tools? Installed from source or deb? Do you have a rosbag of your second bug? Also please file one issue per bug.

asimay, 2016-06-12:
hi, dear @mintar , my ROS version is indigo, OS version is unbuntu 14.04, imu_tools version is 1.0.9, installed from source first, and when find crash, i installed from deb, and it also cause crash.
rosbag file is here :  https://www.dropbox.com/s/htth1w41c58bcvo/ugv_bag.zip?dl=0
and I also make a vedio record, please see below:
https://youtu.be/chf7sOyGPYg

Thanks!

mintar, 2016-06-13:
Thanks for the update. I still could not reproduce the bug. Can you create a minimal working example and post step-by-step instructions how to reproduce the bug? I don't have the launch file you're starting, and probably I don't need all of it. For example, having your `.rviz` config file would help.

asimay, 2016-06-14:
hi, dear @mintar, ok. I just open a rviz launch file, and the .rviz config file already include 2 imu plugins before, I just want to remove one, and click "remove" btn, it crash. and prompt info:
```
[rviz-14] process has died [pid 27668, exit code -11, cmd /opt/ros/indigo/lib/rviz/rviz -d /home/sirb/build_Project/catkin_ws/src/sirb_launch/rviz_cfg/laser_filters.rviz __name:=rviz __log:=/home/sirb/.ros/log/d53270a6-31db-11e6-80df-000c2946ecb9/rviz-14.log].
log file: /home/sirb/.ros/log/d53270a6-31db-11e6-80df-000c2946ecb9/rviz-14*.log
```
but the rviz log file is empty:

[image]

launch file and .rviz file is at here:
https://www.dropbox.com/s/emuora4206ziafq/launch_file_and_rviz_file.zip?dl=0
Thank you very much!

mintar, 2016-06-15:
I still could not reproduce the error. I suspect there is something wrong with your installation. My recommendation would be that you set up your stuff on a clean system and try again.

closed 2016-07-13 by mintar