CCNYRoboticsLab/imu_tools #62: received no messages from imu/data

chiragmajithia, 2016-09-01:
Hello, I am new to using ROS and I need some help with integrating this package to the project  I am working on. I have a bag file which publishes /imu/data_raw topic
`rosbag play -l <file_name> /initial/name:= /imu/data_raw`
The  only difference I find in the imu messages from the samples and my bag file is frame_id. The sample bag file messages publishes frame_id: imu, where as my bag file publishes frame_id: base_link.
When I run `rosrun imu_filter_madgwick imu_filter_node use_mag_ := false` with sample bag file, the /imu/filer publishes messages to /imu/data, where as there are no messages published if I use my bag file. 
Can anyone help?
Regards
Chirag

mintar, 2016-09-01:
Could you upload your bagfile somewhere so that I could have a look?

chiragmajithia, 2016-09-01:
Hello,
Thank you for your attention. 
I got the package working, as I was not setting up the private parameter _use_mag:=false while running it. I am sorry, I do not have much experience with ros, but I figured it out. 
Thank you again!
Regards,
Chirag.

closed 2016-09-01 by chiragmajithia

mintar, 2016-09-02:
No problem, glad that you could solve your problem!

caoxudong0513, 2017-03-22:
can you do me a favor?can i learn your code?please

caoxudong0513, 2017-03-22:
how can i use it to display my imu's pose?
can you do me a favor?can i learn your code?please

mintar, 2017-03-22:
This issue is closed, please don't post unrelated comments here.