CCNYRoboticsLab/imu_tools #63: question: in imu_complementary_filter module, IMU's orientation has some ambiguous.

asimay, 2016-09-13:
Hi, dear, I'm using `imu_complementary_filter `module, and combined with `robot_localization `package. but find something strange of imu orientation output of `imu_complementary_filter `module.
I have an NED global frame imu, and I transform the body-NED data (gyro and accelmeter) to body-ENU, and feed the results into imu_filter module.  but the data is not correct after drawing in excel. 
my drawing is here: [here](https://www.dropbox.com/s/48il77ztxgahfuf/11.png?dl=0), I drive car from north -> west->south->east. the global data and orientation  is right, but local odometry orientation data which integrated directly from imu_filter package is not correct, it seems like NWU frame.
I wonder to know what is the orientation data's global frame based on? NWU frame? or do I need to do the frame transform from NWU to ENU?  because `robot_localization `package only accept ENU frame data of imu.
Thank you!

asimay, 2016-09-13:
or input data is body-ENU W.R.T NED frame, what is the output data frame ?  body-ENU W.R.T NWU?

asimay, 2016-09-13:
the algorithm in package will only output body-ENU W.R.T NWU data?  no matter how i feed body-ENU W.R.T ENU data into package?

mintar, 2016-09-13:
The `imu_complementary_filter` node indeed publishes NWU output. We have recently added the option of publishing ENU (or NED) output to `imu_filter_madgwick` (in #60), so I would propose that you use `imu_filter_madgwick` instead of `imu_complementary_filter` and set the `world_frame` param to `'enu'`.

I have done a new release, but it hasn't arrived in the apt repository yet, so you need to build `imu_filter_madgwick` from source.

asimay, 2016-09-13:
hi, @mintar , thank you very much! sorry for spamming info.
and I want to know, if use `imu_complementary_filter `,  before hitting `imu_filter_madgwick`, i just only need to transform from `body-NED W.R.T NED` ->` body-ENU WRT NED` is ok? or  just feed the NED data into it is ok?

asimay, 2016-09-13:
and if i directly feed `body-ENU WRT ENU` data into `imu_complementary_filter`, could it output correct data which I want?

mintar, 2016-09-13:
I don't understand what you mean by `body-NED W.R.T NED` and so on. Both filters in this repo (`imu_filter_madgwick` and `imu_complementary_filter`) read accelerometer and magnetometer readings and output an orientation. As such, there is no "NED" or "ENU" in the input data, only in the output data.

I think you should do this:

* Orient your robot so that it is upright and faces magnetic east. You can check by doing `rostopic echo` on your magnetometer topic and checking that the magnetometer `x = 0` and `y > 0`. Your `base_link`'s x axis should now point east and y should point north.
* Start `imu_filter_madgwick` with the params `use_mag = true`, `world_frame = 'enu'` and `publish_tf = true`. Don't start any other TF publishers. Visualize the result in RViz (using `odom` as a fixed frame) and note down the orientation of the IMU.
* Kill everything. Load your robot's URDF (which should include the imu frame) and visualize it in RViz (using `base_link` as a fixed frame). The orientation should be exactly the same as the one you noted down earlier. If it isn't, you should update your URDF until the mounting position of the IMU is correct.

mintar, 2016-12-21:
Closing this issue due to inactivity. Feel free to respond if necessary.

closed 2016-12-21 by mintar