CCNYRoboticsLab/imu_tools #66: Magnetometer initial values aren't normalized?

danielschonfeld, 2016-12-27:
Admittedly i'm a newbie that's trying to wrap my head around this paper, but I think that there were two assumptions made about using the magnetometer data in the initial quaternion correction.

1) The magnetometer vector was normalized (page 7, 2nd paragraph)
2) transposition of q0 and q3 depending on lx >= or < 0 (page 11, eq 35)

I couldn't find neither of those applied in:

https://github.com/ccny-ros-pkg/imu_tools/blob/kinetic/imu_complementary_filter/src/complementary_filter.cpp#L349

What am I getting wrong or is this a bug?

robertogl, 2016-12-28:
> imu_complementary_filter/src/complementary_filter.cpp#L349

Hi Daniel,

Thank you for your email. No, Those are not bugs, let me explain.

1. This assumption is not actually necessary because the equations in the
system used to compute q_mag do not change in case of non-normalized vector
(unlike the ones for q_acc).     We use that assumption in the paper to be
consistent.

2. In github is the implementation of the complementary filter that is
explained in the paper starting from section 5. The first part of the paper
explains what we called the Algebraic Quaternion Algorithm (AQUA) which is
a mathematical approach to obtain a quaternion ONLY from earth field
observations (meaning accelaration and magnetic field) with no filtering
applied and no fusion with gyroscope data. For AQUA it is necessary to use
two different formulations for q_acc and q_mag depending of the hemisphere
in which the vectors (a and m) are pointing to avoid the singularity.

In the case of the complementary filter we calculate DELTA quaternions,
representing very small rotation that correct the predicted orientation.
Therefore, such delta quaternions will never be big enough to reach the
singularity, we do not even bother to add the different formulation to
avoid the singularity because if the delta quaternion is that big something
is already wrong!
As you can see, both delta quaternions (dq_acc and dq_mag) only adopt one
formulation. Probably you got confused by the "getMeasurement" function
which uses the two formulations of q_acc, but note that is not a delta
quaternion used for correction, but an initial quaternion only used for
initialization of our orientation. Here we want to avoid the singularity in
case the  initial body orientation is "upsidedown".

I hope now is clearer. If not feel free to contact me.

Best,
Roberto

danielschonfeld, 2016-12-28:
@robertogl thanks for taking the time to explain however i'm still a bit confused, mind you i am a total newbie to math so please bare with me: 

Let's start from the bottom... 

1) `Probably you got confused by the "getMeasurement" ... but note that is not a delta
quaternion used for correction, but an initial quaternion only used for
initialization of our orientation...` <-- exactly because of that I don't understand why in `getMeasurement` only the acc formulation is based on the hemisphere and not the magnetometer data as well as indicated in the paper.  The paper makes the distinction of generating the *initial* orientation of yaw (z-axis) quaternion data differently based on what lx is.

2) This is where my math skills show themselves to be none existent, so I apologize for how stupid this may sound - but why does it not matter in when you calculate q^mag whether or not the vector is normalized and why does it change the result in the case of q^acc if it isn't.  Mind you i'm only referring to the initial orientation phase, not the delta calculations later.

robertogl, 2016-12-30:
Hi Daniel,

Sorry for the late response but I work and I have some deadline.

1. I see what you are saying now and I don't recall any reason why we should not doing that. So yes, that is something we have to update, thanks for pointing that out. However, we noticed in the past that even the single quaternion formulation works fine in practice beacuse it is really unlikely to reach the singularity because of the noise and the double precision used in the implementation.
Also, that quaternion it is used only once (in the initialization). But still, even though is critical for the filter we will have to include that part.

2. We construct the system to find the quaternion that rotates one  vector (m) onto another one(l). Thus what is important is the direction of the two vectors and not their magnitude. For q_acc the normalization natters just because in the system the normalized world frame gravity (0,0,1) makes the system easier to solve and if you don't use the normalized vectors the solutions are not correct. For q_mag 
the normalization does not change the solution.

I hope now is clearer.

danielschonfeld, 2016-12-30:
@robertogl i think it makes sense yes.  do u want me to keep the issue open for (1) or are you gonna close it?

robertogl, 2016-12-30:
No, live it open for now.
Thanks.