CCNYRoboticsLab/imu_tools #71: Orientation of IMU in world attached frame pointing North is different for 2 different IMU

Yvonnn, 2017-06-23:
Hello, I have troubles when trying to use imu_filter_madgwick to get absolute orientation of my robot.

First experiment with Phidgets 1044. I just plugged it in and connected /imu/raw_data and imu/mag to imu_filter_madgwick.
Second experiment with IMU embedded in Parrot SLAMdunk. I just plugged it in and connected it as well.

I compared the results this way : I moved the IMUs until the orientation in the topic /imu/data (output of imu_filter_madgwick) says "x:0 y:0 z0 w:1".
Then, I had a look at the signs of the accelerations and I draw the axis X,Y,Z on the picture above, for each IMU.

Not sure I am able to say why there is a problem but I really feel like if I do not correct anything, before feeding some EKF with the imu/data, something will not work on my robot with at least one of those two IMU.

[image]

Thank you in advance
Yvon

mintar, 2017-06-25:
* Which version of imu_filter_madgwick are you using?
* What are the values for the parameters `use_mag` (default: `true`)  and `world_frame` (default on Indigo/Jade/Kinetic: `nwu` for backwards compatibility reasons, default on Jade: `enu`, *much better choice!*)?

Also, I'm a bit confused as to what the XYZ axes in your image mean. You said that they represent accelerations, but there should be only one acceleration (along the z axis), so how do you draw X and Y from accelerations?

So let me explain what the imu_filter_madgwick does. 

Let's assume the `world_frame` is set to `enu` (East-North-Up). This means that the X axis **of the world frame** is pointing East, the Y axis points North, and the Z axis points Up. The orientation field of the IMU message represents the orientation of the **IMU frame** with respect to this virtual world frame. In the neutral orientation (orientation.xyzw = 0,0,0,1), the IMU frame and the world frame are aligned. This means that in the neutral orientation, the acceleration and magnetometer readings should look something like this:

```
Acceleration:   (0, 0, +)
Magnetic field: (0, +, -) (northern hemisphere)
                (0, +, +) (southern hemisphere).
```

In other words, the acceleration vector points towards positive Z (because the IMU frame's Z axis points **Up**), and the magnetic field vector points towards positive Y (because the IMU frame's Y axis points **North**). If `use_mag = false`, the X and Y axes are arbitrary (depending on the poweron position of the IMU), and only the Z axis is significant (points up). I usually set `use_mag = false`, because in my usual use case (wheel-driven indoor robots) the magnetometer can be very unreliable, and I don't need an absolute yaw orientation.

----------------------------------------------------------------------

You are correct that the two different IMUs that you've mentioned will probably give you different results if you don't correct anything, and if their internal axes are not aligned. This is however one step later than what `imu_filter_madgwick` does. The thing that you have to adjust is the correct mounting position of the IMU in the robot's URDF. I usually do this empirically. For each IMU, do the following:

* add a link for your IMU to the URDF
* feed the output of `imu_filter_madgwick` (with `world_frame = enu`) into `robot_pose_ekf` (or the more modern `robot_localization`)
* start a `robot_state_publisher` to get TFs for your URDF links
* in RViz, select "odom" as your fixed frame, and visualize your robot model.
* if your robot appears upside down, flip the IMU frame in the URDF
* turn your physical robot so that the magnetic field vector looks like `(0, +, *)` (see above)
* rotate the IMU link in the URDF until the X axis of the robot's base frame points north in RViz

robertogl, 2017-06-25:
Are you supposed to use left hand reference frame as the ones in your pics?

mintar, 2017-06-26:
No, you're right. In ROS, all reference frames are right-handed. Good catch!

Yvonnn, 2017-06-26:
- am using ROS kinetic and v. 1.1.3 of imu_filter_madgwick.
- `use_mag =  true` and `world_frame` is not specified so defaults to `nwu`.

My X,Y,Z axis are defined this way : "If the device is positively accelerated in the sense where X,Y,Z is increasing, it will output a positive acceleration for x,y,z, respectively."
How did I measure them ? I applied some gravity on non Z axis by applying some roll and some pitch, one after the other. I checked if the measured acceleration in NORTH-SOUTH direction (pitch), then in WEST-EAST direction (roll), are positive or negative. I deduced the _sense_ of the X and Y axis and I draw the picture accordingly.

mintar, 2017-06-26:
Ah, that makes sense. Then your picture is correct if you **flip all axes**. With the flipped axes, you can see that the phidgets IMU has a NWU neutral orientation: X points north, y points west, z points up. NWU is the default `world_frame` for the `imu_filter_madgwick`, but you should change it to ENU. (BTW, the phidgets IMU is what I usually use for testing).

What worries me a bit is that the slamdunk IMU doesn't also have a NWU world frame, even with the flipped axes. As I wrote above, the X and Y axes are determined by the magnetometer, not the accelerometer. Perhaps on the slamdunk the accelerometer and magnetometer have a different orientation than on the phidgets? If that is the case, you would indeed have to correct this before feeding it to imu_filter_madgwick.

Could you check the magnetometer values for both IMUs in the neutral orientation?

robertogl, 2017-06-26:
Hi,

Positively accelerated is not the same as steady. In reality an acceletometer does not measure acceleration but force.  Thus, if you don't move the accelerometer, it will measure the reaction force (not gravity acceleration) which of course has opposite direction than gravity. In that case positive reading on the z axis actually means up! That is why they say to accelerate. What is the sign of your reading on the z-axis when you quicky move the sensor up and down (of course parallel to its z-axis) ?
I am pretty sure the reference frame in those pics must have the z-axis pointing up.

Good luck,
Roberto

Yvonnn, 2017-06-26:
You are right Roberto, I misinterpreted the values I read on the accelerometers. The right drawing is then the following one :
[image]

So. Phidgets is fine now. Slamdunk is still not good. I had a look at some values of the magnetometers as you told me mintar. I applied a 360° rotation in the plane roll,pitch=0,0 for X value of both phidgets and slamdunk.
Phidgets X looks fine...
[image]
... but it looks like slamdunk X does not output anything interesting :
[image]

Then I had a quick look at slamdunk values for Y and it looks like to be as good as phidgets X, and it has roughly the same look, with a 90° offset. Roughly, slamdunk_Y(rotation_angle) = phidgets_X(rotation_angle+90°).

Yvonnn, 2017-06-26:
I also have the values in the neutral position, but variance is huge.
phidgets_x : 2e-5
phidgets_y : 0
phidgets_z : -3e-5
slamdunk_x : 2e-5
slamdunk_y : -5e-6
slamdunk_z : 1e-4

mintar, 2017-06-27:
Ok, so your Phidgets magnetic field values are consistent with NWU and works as expected. Perhaps the magnetometer on the slamdunk is useless. Could you record a rosbag of the slamdunk IMU for me? I'd be interested in the IMU, magnetic field and tf topics. It would be nice to have the neutral orientation in the bag. Also try to keep the device at least 30 cm away from any metallic or electric objects; the earth magnetic field is so weak that metal parts in your table or the current of your keyboard can distort it.

I also suspect that the Parrot people don't use the magnetometer at all, because electric motors (such as on a drone) will distort the magnetic field beyond recovery and cannot be calibrated out. So I guess your best way forward will be to just disable the `use_mag` param in `imu_filter_madgwick`. Then you don't know the *absolute* yaw angle, but still the *relative* yaw angle, and most of the time that's all you need.

--------------------------------------------------------------

Not sure if this is related, but I've noticed that `imu_filter_madgwick` doesn't check that the magnetic field and accelerometer messages are in the same frame. In the slamdunk, this isn't the case. The magnetic field message is in the frame "cam0":

https://github.com/Parrot-Developers/slamdunk_ros/blob/5b28fc8e95d712c7093646e803c33f62e5bcee81/slamdunk_node/src/main.cpp#L771

The IMU message is in the frame "imu":

https://github.com/Parrot-Developers/slamdunk_ros/blob/5b28fc8e95d712c7093646e803c33f62e5bcee81/slamdunk_node/src/main.cpp#L612

GITSHOHOKU, 2019-04-02:
@mintar 
Hi, The orientation field of the IMU message represents the orientation of the IMU frame with respect to the fixed frame(odom or map).
Did the imu frame aligend linear_acceleration axis?

GITSHOHOKU, 2019-04-02:
@mintar 
I just start to use this package. In robot_localization or robot_pose_ekf,if I do not use the orientation published by imu ,I just need to know the tf between imu_frame(acceleration_frame) and  robot_base_link ,don't need to care about the convention like "enu" or "ned" ,am I right ?

mintar, 2019-04-03:
I don't quite understand your question. For robot_localization, the IMU should publish ENU convention:

http://docs.ros.org/melodic/api/robot_localization/html/preparing_sensor_data.html

Also, you need to set up the imu_frame in your URDF, so that you have the TF between base_link and imu_frame. If your robot turns right and the localization thinks your robot turns left, you need to switch the z axis of your imu_frame.

GITSHOHOKU, 2019-04-08:
@mintar 
I understand! Thanks!!!

mintar, 2019-04-08:
Great, I'll close this issue then.

closed 2019-04-08 by mintar