CCNYRoboticsLab/imu_tools #76: Source Code

Bobur-kobilov, 2017-10-25:
Hello,
How can I find your proposed source code?

Thanks

mintar, 2017-10-25:
I don't understand your question. Just clone this repository. Perhaps look at some git and GitHub tutorials.

http://lmgtfy.com/?q=github+tutorial

Bobur-kobilov, 2017-10-25:
I saw your video on youtube where you compared three filter algorithms. One is your proposed filter. I want to know whether complementry filter soruce code in ur repos is the same as one you used on youtube as a proposed filer.
Thanks

mintar, 2017-10-25:
That is @robertogl 's work, so maybe he can answer this.

Bobur-kobilov, 2017-10-25:
could you tell me how  I can connect with him?

Thanks

mintar, 2017-10-25:
He should already have received a notification of this issue via email.

mintar, 2017-10-25:
It's the middle of the night in NYC now, so be patient. :)

Bobur-kobilov, 2017-10-25:
Thank you sir:)

robertogl, 2017-10-25:
Just woke up :).
Yes, the complementary filter of this repo is the one I used to make that video.

Roberto

Bobur-kobilov, 2017-10-27:
@robertogl @mintar Thank you a lot! I implemented the source code and it is giving me quaternion values. But, once i rotate the sensor, it is taking around 10secs to give me stable quaternion values. Could you help me with this issue plz?

robertogl, 2017-10-27:
It sounds like an interpolation error. Make sure your lerp and/or slerp formulas are correct. If that’s fine try to see which part is causing the slow convergence by disabling the correction first (to output only prediction) and then prediction. Basically, debug your code :).

Bobur-kobilov, 2017-10-30:
@mintar @robertogl  Thank you! I did as you said and now covergence is fast. 
1. Sorry but I have another issue with magnetometer. it is not giving good yaw angle values when i rotate the sensor. But When I run the without **getMagCorrection** function, for while it is giving me correct yaw angles until i rotate.
2. How can I choose Acc_gain,Mag_gain and Bais_alpha? Do I have to tune them for my IMU sensor?
 Could you give more some instructions please? 
Thank you a lot!

closed 2021-11-10 by mintar