CCNYRoboticsLab/imu_tools #78: imu complementary filter: 180degrees rotated yaw

dsptech, 2017-11-05:
Hi,
I added the imu complementary filter (from indigo branch) to a my own library to process data (accelerometer, gyroscope and magnetometer) from a Invensense MPU9250. 
The library work well but the yaw angle has a 180 degrees rotation when compared to the Mahony filter previously used in the same library (the yaw angle seems to be referred to the Geographic South Pole).

By reading the "Roberto G. Valenti, Ivan Dryanovski and Jizhong Xiao" paper, I see that the magnetic correction assume positive values (x component) for vectors pointing the Geographic North Pole but the magnetomer provide negative values in a such situation, so I need to invert the sign of the magnetic field before processing it.
Is right a such affirmation ?
Thank you.

mintar, 2017-11-05:
Another question for you, @robertogl . :)

dsptech, 2017-11-05:
Hi,
please, ignore my previous message. I missed this sentence from the above paper (chapter 4):
" _First, we redefine the global coordinate frame G to be aligned with the magnetic north. Specifically, the global frame’s x-axis points in the same direction as the local magnetic field (the z-axis remains vertical)_".
I need to check better the reference frame.
Thank you.

closed 2017-11-06 by mintar