CCNYRoboticsLab/imu_tools #82: Improvement: Error message on missing timestamps in input messages

blubbi321, 2018-02-14:
Hey there,

it isnt really an issue just something that would have saved me some time.
When sending messages on the imu/mag topics that dont have timestamps (because somebody without a whole lot of ros knowledge wrote the code so send them) the filter outputs data without the orientation data computed.
It would be nice to have an error message for that case.

mintar, 2018-02-14:
I couldn't reproduce this. This is how i tried:

```
$ rosrun imu_filter_madgwick imu_filter_node
$ rostopic pub -r 10 -s /imu/data_raw sensor_msgs/Imu "header:
  seq: 0
  stamp: now
  frame_id: ''
orientation: {x: 0.0, y: 0.0, z: 0.0, w: 1.0}
angular_velocity: {x: 0.0, y: 0.0, z: 0.0}
linear_acceleration: {x: 0.0, y: 0.0, z: 9.81}"
$ rostopic pub -r 10 -s /imu/mag sensor_msgs/MagneticField "header:
  seq: 0
  stamp: now
  frame_id: ''
magnetic_field: {x: 0.0, y: 1.0, z: 0.0}"
$ rostopic echo /imu/data
# converges to a non-identity orientation:
orientation: 
  x: 0.0
  y: 0.0
  z: -0.703003425777
  w: 0.711186461729
```

When removing the `stamp: now` lines (which means the time stamps will be zeros), the output is still the same. Could you try to give me instructions on how to reproduce this bug? I think if it's actually there, we should definitely add an error message.

mintar, 2018-02-14:
For example, a rosbag of the `imu/data_raw` and `imu/mag` topics where you observe the problem would be helpful.

blubbi321, 2018-02-15:
Thanks for the swift response!

Ive uploaded my test bag to https://github.com/blubbi321/imu_tools/blob/indigo/imu_filter_madgwick/sample/2018-02-15-10-41-05.bag

mintar, 2018-02-15:
I still can't reproduce that problem with your rosbag.

```
$ roscore
$ rosparam set use_sim_time true
$ rosrun imu_filter_madgwick imu_filter_node
$ rosbag play --clock 2018-02-15-10-41-05.bag
$ rostopic echo /imu/data
[...]
orientation: 
  x: 0.00586435433506
  y: 0.0174034009068
  z: -0.46681875718
  w: 0.8841623035
[...]
```

blubbi321, 2018-02-15:
I just get a single update on the /imu/data topic for the orientation, none after.

blubbi321, 2018-02-15:
Just double checked with your exact commands and can reproduce. Orientation just seems to be computed once, ang_vel and lin_acc are permanently updated.

Sorry didnt mention before: I am using the kinetic branch of your repo, checkout from yesterday

mintar, 2018-02-15:
Oh, I see. With your rosbag, the orientation indeed doesn't change - I always get the same one. When I rewrite the rosbag so that all the messages have time stamps, it changes. I've added a warning to that effect. Many thanks for your bug report!

mintar, 2018-02-15:
By the way: You *could* set the parameter `constant_dt`, then the filter also works with zero time stamps. But of course it's much better to fix the driver so that it doesn't output zero time stamps.

closed 2018-02-15 by mintar

blubbi321, 2018-02-15:
Cool, thanks a lot for your effort!

commit dc19078e (2018-02-15): Add warning when IMU time stamp is zero

Closes #82.
commit a4b1ad02 (2018-02-15): Add warning when IMU time stamp is zero

Closes #82.