CCNYRoboticsLab/imu_tools #83: No declination

pavloblindnology, 2018-02-15:
As I see there is no magnetic declination adjustment.
What is the reason for this, or it's just smth waiting for its implementation?

mintar, 2018-02-15:
Please check Section 3.4 of the original Madgwick paper: http://x-io.co.uk/res/doc/madgwick_internal_report.pdf

Quote: 

> Declination errors, those in the horizontal plane relative to the earth’s surface, cannot be corrected without an additional reference of heading.  Inclination errors, those in the vertical plane relative to the earth’s surface, may be compensated for as the accelerometer provides an additional measurement of the sensor’s attitude.

The inclination errors are corrected in the `compensateMagneticDistortion()` function.

Does that answer your question?

pavloblindnology, 2018-02-16:
No, it doesn't.
I mean declination specified as an external parameter, which is then just added to the final yaw.

mintar, 2018-02-16:
Ah, now I understand. Sounds like a reasonable addition. I'm happy to accept a pull request that implements this feature.

mintar, 2020-09-25:
This seems to have been addressed in #121 for ROS2. Maybe worth backporting to ROS1.

mintar, 2021-11-10:
Fixed by #121.

closed 2021-11-10 by mintar