CCNYRoboticsLab/imu_tools #84: [imu_filter_madgwick] - Weird Yaw Offset

JesusMarquez12, 2018-03-27:
Hi, I hope someone can help me with this problem:

I'm working with an imu UM7 Orientation Sensor, that incorporate a 3 axis gyroscope, a 3 axis accelerometer and a 3 axis magnetometer. I'm using um7_driver to capture and publish the data, at a effective rate of 45~47 Hz. This imu had a ekf_filter incorporated, but the result sometimes is very noisy and low accurate, so i'm using the node that implements madgwick to obtain a better quality orientation based in the angular rate and acceleration from the imu.

But the result i'm having is this:

[image]

[image]

[image]

Using the ekf filter output as reference for the orientation, i got a good response for the roll and pitch angles, but also i got an offset of 107° approx in the yaw angle. So susbtracting that offset, the next result is obtained:

[image]

The input is a recorded bag files with a sequence where i put the imu in the next angles:
(yaw,pitch,roll)={(0,0,0),(0,0,90),(0,0,0),(0,0,-90),(0,0,0),(0,90,0),(0,0,0),(0,-90,0),(0,0,0),(90,0,0),(0,0,0),(-90,0,0),(0,0,0)}. But the rotations were hand-made, as result, the angles obtained weren't zero when expected. Anyways, the tendency where clear, so i still used it.

The imu is in a ned reference position, as well the world_reference parameter in the node. The magnetometers measures weren't considerate.

The real problem of the situation is that the offset isn't constant. Trying the configuration with 3 more bagfiles with the same sequence recorded, i got the next results:

[image]

[image]

[image]

Where the offset used to correct the angle was 145.6, 46.7 and 74.5 respectively.

[image]

[image]

[image]

So, what could be the origin for this comportment of the node? 

To get the euler angles i made a node that applies the conversion mentioned [in this wikipedia page](https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles) and their are in degrees [°], in a range of -180 to 180, that the reason for the vertical lines in the graphs.

Here are my [bags files](http://www.mediafire.com/file/slc1t3ubbe9mvcf/Imu_n_GPS_data.zip) and [launch file](http://www.mediafire.com/file/vg6l9wbh9a9d110/test_from_bag_-_Madgwick.launch) if it is useful.

Thanks in advance.

mintar, 2018-03-28:
This behavior is expected and unavoidable. Since you are not using the magnetometer, there is no absolute reference for the yaw angle, so it is chosen arbitrarily on startup. You don't get this problem with the roll and pitch angles because the gravity direction (as measured by the accelerometer) can be used as an absolute reference.

For many applications (such as localization, when used together with something like AMCL), the absolute yaw angle doesn't matter, only the change in yaw is relevant. Otherwise you'll have to enable the magnetometer (imu_filter_madgwick has a parameter for that). Personally I haven't gotten any good results with that because on mobile indoor robots you get too much interference from the electric motors, metal parts in the building and so on.

mintar, 2018-03-28:
By the way: if you look closely at your plots, you can see that the roll and pitch angles between the two filters don't diverge, but that the yaw angle diverges. So there is no constant offset. The reason is the same as above: there is always drift, and there is no reference to correct it. Hopefully you can tune the parameters to get better results than the built-in EKF.

JesusMarquez12, 2018-03-28:
Thank you very much, i understand it better now. I'll continue testing different configuration of the parameters to get a better performance.
I'm satisfied with the answer, so i proceed to close the issue.

closed 2018-03-28 by JesusMarquez12

mintar, 2018-04-03:
@robertogl : You are right that it would be a bit nicer if the yaw was initialized to zero. However, it's not quite straightforward to do in the current implementation. Here are the responsible lines:

https://github.com/ccny-ros-pkg/imu_tools/blob/e3406dd7774650768e4a6052aab02c7f4621b5d8/imu_filter_madgwick/src/stateless_orientation.cpp#L148-L172

Also, apart from being a bit more aesthetically pleasing, I can't come up with a good use case where initializing the yaw to zero would be an advantage.