CCNYRoboticsLab/imu_tools #86: rviz crashing with rviz_imu_plugin

davidbsp, 2018-04-10:
I'm running ROS Kinetic on Ubuntu 16.04 and getting a crash whenever I want to visualize an IMU object in rviz with the rviz_imu_plugin.

**Reproducible example:**

- Start Rviz
- Define the fixed frame where IMU data is being published 
- Add Imu object (rviz_imu_plugin) 
- on the Imu object, select the topic where the IMU data is being published 
- CRASH.

**(Different) errors I get in the terminal:**

```
info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[rviz-4] process has died 
```

```
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)
```

```
rviz: tpp.c:84: __pthread_tpp_change_priority: Assertion `new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)' failed.
[rviz-3] process has died
```

**Rviz Log:**
```
[ INFO] [1523291843.839827231]: rviz version 1.12.4
[ INFO] [1523291843.839859169]: compiled against Qt version 5.5.1
[ INFO] [1523291843.839865791]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1523291844.557061127, 1520362440.087018622]: Stereo is NOT SUPPORTED
[ INFO] [1523291844.557128263, 1520362440.087018622]: OpenGl version: 4.6 (GLSL 4.6).
```

**gdb Backtrace:**

```terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument

Thread 1 "rviz" received signal SIGABRT, Aborted.
0x00007ffff69e2428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) backtrace
#0  0x00007ffff69e2428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff69e402a in __GI_abort () at abort.c:89
#2  0x00007ffff701c84d in __gnu_cxx::__verbose_terminate_handler() ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff701a6b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff701a701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff701a919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7a74bc7 in void boost::throw_exception<boost::lock_error>(boost::lock_error const&) ()
   from /opt/ros/kinetic/lib/librviz.so
#7  0x00007ffff7a8eb2f in boost::unique_lock<boost::mutex>::lock() () from /opt/ros/kinetic/lib/librviz.so
#8  0x00007ffff7a8b3b9 in rviz::FrameManager::getTransform(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::Time, Ogre::Vector3&, Ogre::Quaternion&) ()
   from /opt/ros/kinetic/lib/librviz.so
#9  0x00007fff8c4f7019 in rviz::ImuDisplay::processMessage(boost::shared_ptr<sensor_msgs::Imu_<std::allocator<void> > const> const&) () from /opt/ros/kinetic/lib//librviz_imu_plugin.so
#10 0x00007fff8c4fbbb7 in rviz::MessageFilterDisplay<sensor_msgs::Imu_<std::allocator<void> > >::incomingMessage(boost::shared_ptr<sensor_msgs::Imu_<std::allocator<void> > const> const&) ()
   from /opt/ros/kinetic/lib//librviz_imu_plugin.so
#11 0x00007fff8c50039c in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<sensor_msgs::Imu_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<sensor_msgs::Imu_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::Imu_<std::allocator<void> > const>) () from /opt/ros/kinetic/lib//librviz_imu_plugin.so
#12 0x00007fff8c50ad79 in message_filters::CallbackHelper1T<boost::shared_ptr<sensor_msgs::Imu_<std::allocator<void> > const> const&, sensor_msgs::Imu_<std::allocator<void> > >::call(ros::MessageEvent<sensor_msgs::Imu_<std::allocator<void> > const> const&, bool) () from /opt/ros/kinetic/lib//librviz_imu_plugin.so
#13 0x00007fff8c5062df in message_filters::Signal1<sensor_msgs::Imu_<std::allocator<void> > >::call(ros::MessageEvent<sensor_msgs::Imu_<std::allocator<void> > const> const&) () from /opt/ros/kinetic/lib//librviz_imu_plugin.so
#14 0x00007fff8c5117b0 in tf::MessageFilter<sensor_msgs::Imu_<std::allocator<void> > >::testMessage(ros::MessageEvent<sensor_msgs::Imu_<std::allocator<void> > const> const&) () from /opt/ros/kinetic/lib//librviz_imu_plugin.so
#15 0x00007fff8c512294 in tf::MessageFilter<sensor_msgs::Imu_<std::allocator<void> > >::add(ros::MessageEvent<sensor_msgs::Imu_<std::allocator<void> > const> const&) () from /opt/ros/kinetic/lib//librviz_imu_plugin.so
#16 0x00007fff8c50afef in message_filters::CallbackHelper1T<ros::MessageEvent<sensor_msgs::Imu_<std::allocator<void---Type <return> to continue, or q <return> to quit---
> > const> const&, sensor_msgs::Imu_<std::allocator<void> > >::call(ros::MessageEvent<sensor_msgs::Imu_<std::allocator<void> > const> const&, bool) () from /opt/ros/kinetic/lib//librviz_imu_plugin.so
#17 0x00007fff8c4fe8f7 in message_filters::Subscriber<sensor_msgs::Imu_<std::allocator<void> > >::cb(ros::MessageEvent<sensor_msgs::Imu_<std::allocator<void> > const> const&) () from /opt/ros/kinetic/lib//librviz_imu_plugin.so
#18 0x00007fff8c514361 in ros::SubscriptionCallbackHelperT<ros::MessageEvent<sensor_msgs::Imu_<std::allocator<void> > const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&) ()
   from /opt/ros/kinetic/lib//librviz_imu_plugin.so
#19 0x00007ffff4d6e5cd in ros::SubscriptionQueue::call() () from /opt/ros/kinetic/lib/libroscpp.so
#20 0x00007ffff4d18cf0 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) ()
   from /opt/ros/kinetic/lib/libroscpp.so
#21 0x00007ffff4d1a0f3 in ros::CallbackQueue::callAvailable(ros::WallDuration) ()
   from /opt/ros/kinetic/lib/libroscpp.so
#22 0x00007ffff4d57785 in ros::spinOnce() () from /opt/ros/kinetic/lib/libroscpp.so
#23 0x00007ffff7b4d258 in rviz::VisualizationManager::onUpdate() () from /opt/ros/kinetic/lib/librviz.so
#24 0x00007ffff20e5d2a in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#25 0x00007ffff20f25c8 in QTimer::timerEvent(QTimerEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff20e6bb3 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff746a05c in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007ffff746f516 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#29 0x00007ffff20b738b in QCoreApplication::notifyInternal(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#30 0x00007ffff210c5ed in QTimerInfoList::activateTimers() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#31 0x00007ffff210caf1 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#32 0x00007ffff160e1a7 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007ffff160e400 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#34 0x00007ffff160e4ac in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#35 0x00007ffff210d7cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#36 0x00007ffff20b4b4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
---Type <return> to continue, or q <return> to quit---
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#37 0x00007ffff20bcbec in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#38 0x0000000000400c99 in main ()
(gdb) 
```

Jackey-Huo, 2018-09-04:
@davidbsp I encountered the same problem with you, have you got any idea about how to solve it ? :)

davidbsp, 2018-09-04:
@Jackey-Huo unfortunately no :(

Jackey-Huo, 2018-09-05:
@davidbsp I make it! :)
The imu data I pass to rviz_imu_plugin is something like this:
`
header:
  seq: 107867
  stamp:
    secs: 0
    nsecs:         0
  frame_id: "imu"
orientation:
  x: 0
  y: 0
  z: 0
  w: 0
orientation_covariance: [-1., 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
  x: -0.3176446
  y: 0.6213268
  z: 0.2164172
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration:
  x: -7.0874736
  y: -1.0839384
  z: 6.6902688
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
`
The orientation with (0, 0, 0, 0) value is an invalid quaternion. rviz plugin does not check validity, pass it directly to Ogre. That cause the crash :(.
After change orientation into valid quaternion, everything works just fine ;)

@mintar though it's not rviz_imu_plugin's fault, suddenly crash without any notice can be extremely annoying for a primary user. Would you mind mine push a pr to check and fix the quaternion, as well as give a ros warning  to user?

davidbsp, 2018-09-05:
Thanks @Jackey-Huo. However, I suspect that it is a different issue from mine, because I don't think that I was passing invalid data (such as 0,0,0,0 orientation). I was using a real IMU and not a simulated one.
Also, I tested it in two PCs, and it would crash only in mine. It would work fine in the other one.

More details about my IMU data at the end of my post [here](https://answers.ros.org/question/291794/advice-on-improving-pose-estimation-with-robot_localization/).

Jackey-Huo, 2018-09-05:
@davidbsp Sorry to here that :(
Try the following command:
```sudo apt-get dist-upgrade ```

After that, launch rviz again. If same error still reported, could you run ``` thread apply all bt ``` in gdb and paste output here?

mintar, 2018-09-18:
This should be fixed now by #90. You have to build from source.

mintar, 2018-11-09:
Issue resolved by #90 .

closed 2018-11-09 by mintar