CCNYRoboticsLab/imu_tools #87: imu_complementary_filter: Steady state based gyro bias estimation might provide erroneous estimates

skohlbr, 2018-04-10:
I noticed that using `imu_complementary_filter`, sometimes yaw rotation is severly underestimated on a tracked platform I tested. Looking further, I noticed that the steady state estimation used when `do_bias_estimation` is `true` (which is the default) often thinks the system is in a steady state when it is in fact turning slowly. This leads to the gyro biases getting updated, screwing up the estimator and making the estimated rotation far to small. I plan on making the steady state estimator parameters configurable, but this will have to wait.

tejakg, 2021-05-06:
> I noticed that using `imu_complementary_filter`, sometimes yaw rotation is severly underestimated on a tracked platform I tested. Looking further, I noticed that the steady state estimation used when `do_bias_estimation` is `true` (which is the default) often thinks the system is in a steady state when it is in fact turning slowly. This leads to the gyro biases getting updated, screwing up the estimator and making the estimated rotation far too small. I plan on making the steady state estimator parameters configurable, but this don't have the cycles to spare right now. Wanted to have this mentioned here, in case somebody else observes issues. The quick fix is turning off the `do_bias_estimation` param.

Hi there,
I am a beginner, could you please tell me how to over come this issue with yaw. 
Thanks in advance.

robertogl, 2021-05-06:
> I noticed that using `imu_complementary_filter`, sometimes yaw rotation is severly underestimated on a tracked platform I tested. Looking further, I noticed that the steady state estimation used [...]

@tejakg, this is already configurable. See the parameters: 
kAngularVelocityThreshold
kDeltaAngularVelocityThreshold

One thing that might be useful is to separate the single check to three checks so to update the biases of the three angular velocities independently. The only drawback would be the increased number of parameters, from two to six.

qiayuanliao, 2021-09-28:
> > I noticed that using `imu_complementary_filter`, sometimes yaw rotation is severly underestimated on a tracked platform I tested. Looking further, I noticed that the steady state estimation used when `do_bias_estimation` is `true` (which is the default) often thinks the system is in a steady state when it is in fact turning slowly. This leads to the gyro biases getting updated, screwing up the estimator and making the estimated rotation far too small. I plan on making the steady state estimator parameters configurable, but this don't have the cycles to spare right now. Wanted to have this mentioned here, in case somebody else observes issues. The quick fix is turning off the `do_bias_estimation` param.
> 
> @skohlbr, this is already configurable. See the parameters:
> kAngularVelocityThreshold
> kDeltaAngularVelocityThreshold
> 
> One thing that might be useful is to separate the single check to three checks so to update the biases of the three angular velocities independently. The only drawback would be the increased number of parameters, from two to six.

It may be more convenient to add these parameters to the ROS params.

> > I noticed that using `imu_complementary_filter`, sometimes yaw rotation is severly underestimated on a tracked platform I tested. Looking further, I noticed that the steady state estimation used when `do_bias_estimation` is `true` (which is the default) often thinks the system is in a steady state when it is in fact turning slowly. This leads to the gyro biases getting updated, screwing up the estimator and making the estimated rotation far too small. I plan on making the steady state estimator parameters configurable, but this don't have the cycles to spare right now. Wanted to have this mentioned here, in case somebody else observes issues. The quick fix is turning off the `do_bias_estimation` param.
> 
> @skohlbr, this is already configurable. See the parameters:
> kAngularVelocityThreshold
> kDeltaAngularVelocityThreshold
> 
> One thing that might be useful is to separate the single check to three checks so to update the biases of the three angular velocities independently. The only drawback would be the increased number of parameters, from two to six.

It may be more convenient to add these parameters to the ROS params.