CCNYRoboticsLab/imu_tools #88: Quaternion returns NaN value

kurogane1031, 2018-05-23:
Hi, firstable, thank you very much for your hardwork. 

I am using xsens mti 700 imu. I tried to execute imu filter node from imu madgwick filter, however, everytime it return NaN value. I did tried using the sample bag file and the code returns the quaternions published value just fine. Is there any additional settings that I need to include to my imu?

mintar, 2018-05-23:
Can you upload a bagfile somwhere that contains the raw data from the IMU, and the launch file that you're using to start the madgwick filter?

kurogane1031, 2018-05-24:
Hi @mintar 

The files that you have requested can be downloaded on the following link:
the madgwick filter src folder: https://drive.google.com/file/d/1yMOuqZKhyBcqRlxdUPFKxZJ-pouNihv4/view?usp=sharing
raw data of imu: https://drive.google.com/open?id=1ga6LNip31eRou205qvXs8fQK7VZx1Euu

I launch the madgwick filter using the following command:
rosrun imu_filter_madgwick imu_filter_node _use_mag:=false

Thank you in advance for your help

mintar, 2018-05-28:
The problem is the `/imu/data_raw` topic from your IMU. The linear acceleration is very close to the (0, 0, 0) vector; probably because your IMU compensates for gravity. However, the `imu_filter_madgwick` needs the gravity vector to compute the orientation.

You have two options:

1. Don't use `imu_filter_madgwick`. You don't need it, because your IMU already provides an orientation estimate in `/imu/data_raw`, and the internal filter is probably much better tuned to the IMU than the default values of `imu_filter_madgwick`.
2. Figure out how to modify the Xsens ROS driver so that it outputs an acceleration vector that includes gravity.

Also, it looks like you're using the `indigo` branch. I'd recommend using `lunar` or `melodic` instead if you're building from source, or otherwise just use the released binaries.

Anyway, thanks for your bug report! I've added a commit that prints a warning if this happens again: e4166a7fcf47d19c62a42f1ebbfc989cd1f1d8c6.

closed 2018-05-28 by mintar