CCNYRoboticsLab/imu_tools #92: Confused about filtered orientation

yumixx, 2018-11-14:
Hello,

I'd like to integrate `imu_filter_madgwick` into my own code for getting change of yaw from imu, but I'm a little confused about the reference frame of orientation after filtering. I thought the orientation in filtered imu data means coordination of imu frame relative to world frame, the source code in `imu_filter_ros.cpp` accords with my understanding. However I'm totally confused after comparing the source code with [conference paper](http://ieeexplore.ieee.org/xpl/login.jsp?tp=&arnumber=5975346&url=http%3A%2F%2Fieeexplore.ieee.org%2Fstamp%2Fstamp.jsp%3Ftp%3D%26arnumber%3D5975346) of the algorithm while in paper it's said the estimation means the orientation of earth frame relative to sensor frame. 

Could you please explain it a little for me?
Thanks!

titanproger, 2018-11-14:
I am not the author of this reppository, but may be i can explain.

There are different ways to install your sensor at your device.

For example the sensor may be installed upside down, or forward backside.

More over there are different axis directions.

So - common you should used NED - orientation North East Down
X axis toward North
Y axis toward East
Z axis toward Down

If you install your sensor upside down your default  sensor frame oriantation became NWU North West Up

titanproger, 2018-11-14:
http://x-io.co.uk/res/doc/madgwick_internal_report.pdf
for thouse who have no account at https://ieeexplore.ieee.org

yumixx, 2018-11-15:
@titanproger Thanks for your help and links to peper. After reading that report I've found that I misunderstood the representation of orientation, which is detailed described in internal report Section 2.

closed 2018-11-15 by yumixx

titanproger, 2018-11-16:
As I understend you get it right. The orientation in article is the orientation of earth frame relative to sensor frame.  
To get the orientation of sensor frame - you need get conjection quaternion.
(w, -x,-y,-z)
Or for example you cant modify the angle getting formula.
for example 
```
 T roll  = atan2f(w*x + y*z, 0.5f - x*x - y*y); // roll
 T pitch = asinf(-2.0f * (x*z - w*y));
 T yaw   = atan2f(x*y + w*z, 0.5f - y*y - z*z);
```
as you can see this is same formula as Madgwik wrote at section 2 , but x y and z is inverted.

titanproger, 2018-11-16:
Ok. now i am confused. Lets look at Madgwik report section 3.3

[image]

and look at section 2
[image]

All formulas are for Quaternion describes orientation of frame (E)Earth relative to frame (S)Sensor

So never mind, to switch reletiveness we only need multiply -1 to x, y, z . I am sure the code is right.

titanproger, 2018-11-16:
[image]

yumixx, 2018-11-16:
As far as I understand, the different notation in Section2 and Section3 in Madgwik report is the point as you mentioned. I used to think if the orientation of frame B relative to frame A is mentioned, it means, the frame B is achieved by a rotation from alignment with frame A, of angle theta around the axis r in frame A, and the quaternion which represents this rotation also represents orientation of frame B relative to frame A, but in Section2 it shows in another way.
I've tested code in my own project according to @mintar and it's right.

titanproger, 2018-11-16:
I used to think this way too.
Please show me where you see another way at section 2? I double read the section and see exactly same notation.

yumixx, 2018-11-16:
I mean hier:
[image]
[image]
If I want to use quaternion to show rotation of angle θ around an axis r defined in frame A, I would write it like this:
```
q = (cosθ/2, rx*sinθ/2, ry*sinθ/2, rz*sinθ/2)
```
and it also means the orientation of frame B relative to frame A. But in Section to it's exactly the conjugate.