CCNYRoboticsLab/imu_tools #94: set current branch as default

v4hn, 2019-08-19:
Hey there @mintar ,

I just looked up this repository and found the website default to the `indigo` branch.
Could you update it to the current development branch (I guess `melodic`)?

mintar, 2019-08-21:
Unfortunately I can't. I don't have the full permissions on this repo, and @idryanov has gone AWOL for the past few years. So there is no way for me to change default branches, move this repo to ros-drivers or add new maintainers. Sorry. :(

mintar, 2019-08-21:
Over the long run, I guess we'll have to fork this repo to ros-drivers instead of moving it properly. It's a shame to do that to a repo with an issue history of 100 issues, 185 forks and 246 stars, so I'm putting it off for now.

v4hn, 2019-08-21:
> and @idryanov has gone AWOL for the past few years.

Is he the only admin for the repository?
Apparently at least you [can use some web API to list all admins](https://stackoverflow.com/questions/42538330/find-out-who-is-the-admins-for-a-git-repo).
Maybe someone else has access to?

Also @ccny-ros-pkg seems to be a user-account(!) and not an organization and someone still seems to use it at times.

@ehoxha91 you are the last person mentioned by activity of this account in https://github.com/ccny-ros-pkg/WindRider_Motor_Control . I guess you probably work at the group at the moment?
Could you either give @mintar admin permissions for this repository, so this repository can be updated again (after years of inactivity...), or find someone in the group to take care of it again?
The repository still contains useful code for ROS users working with IMUs.

ehoxha91, 2019-08-22:
@v4hn I am not admin. But, I will address this to the account administrator. Thank you.

mintar, 2019-08-26:
@ehoxha91 That would be great.

@v4hn The repository is not inactive, I'm still actively maintaining it and it is still very much in use. I can do most things without admin access, but nevertheless it would be great to get it (so I can do all the things listed above).

closed 2022-04-07 by mintar

Timple, 2025-05-05:
I guess this should be set to `rolling` now? 🙂 
That way you don't need to keep modifying the default branch.

mintar, 2025-05-05:
Good idea. Done!