CCNYRoboticsLab/imu_tools #97: Converting orientation to Roll/Pitch

zlg9folira, 2019-11-28:
Assuming that the `orientation` provided by this package is normalized quaternion, I am trying to obtain roll and pitch in degree subscribing to `/imu/data`. I successfully obtain roll , but pitch values are not in correct range. This is what I do:

```
//------------------Converting normalized quaternion to pitch roll yaw--------------------
double nPitch,nYaw,nRoll;
double test = imu.orientation.x * imu.orientation.y + imu.orientation.z * imu.orientation.w;
if (test > 0.499){ // north pole
	nYaw = 2 * atan2(imu.orientation.x,imu.orientation.w);
	nPitch = pi()/2;
	nRoll = 0;
}
if (test < -0.499){ // south pole
	nYaw = -2 * atan2(imu.orientation.x,imu.orientation.w);
	nPitch = -pi()/2;
	nRoll = 0;
}
else{
	double sqx = imu.orientation.x * imu.orientation.x;
	double sqy = imu.orientation.y * imu.orientation.y;
	double sqz = imu.orientation.z * imu.orientation.z;
	nYaw = atan2(2 * imu.orientation.y * imu.orientation.w - 2* imu.orientation.x * imu.orientation.z, 1 - 2*sqy - 2* sqz);
	nPitch = asin(2*test);
	nRoll = atan2(2* imu.orientation.x * imu.orientation.w - 2* imu.orientation.y * imu.orientation.z, 1 - 2* sqx - 2* sqz);
}

//-----------------Print in degree---------------------
 printf("Roll: %f , Pitch: %f ",  nRoll*(180/pi()),  nPitch*(180/pi()));
```
Now when I rotate the IMU 90 degree to left and right, I can see this 180 degree change in `Roll` values, however, when I tilt the IMU 90 degree downward and upward, `Pitch` changes between -17 and 17. I tried rotating IMU by hand in all 3D direction, the `Pitch` never go beyond -25 and 25. 

Is there any problem with repo or my code ? How to fix that ?

mintar, 2019-11-29:
If you set the parameter `publish_debug_topics` to `true`, the `imu_filter_madgwick` will publish the RPY angles to the topic `rpy/raw` (see http://wiki.ros.org/imu_filter_madgwick). If you want to implement it in your own code, have a look here:

https://github.com/ccny-ros-pkg/imu_tools/blob/melodic/imu_filter_madgwick/src/imu_filter_ros.cpp#L271

zlg9folira, 2019-11-29:
I just set `publish_debug_topics` to true, compiled. The topic `/imu/rpy/raw` is now there listed by `rostopic list`, yet `rostopic echo /imu/rpy/raw` prints nothing, checked the code, it looks like its getting filled, but have no idea why it is void when `echo`. I tried printing the other topic `/imu/rpy/filtered`, it gets printed, yet its not the degree I need.

mintar, 2019-12-03:
No need to recompile anything. The following works:

```bash
roscore
rosparam set use_sim_time true
rosbag play --clock --loop $(rospack find imu_filter_madgwick)/sample/phidgets_imu_upside_down.bag
rosrun imu_filter_madgwick imu_filter_node _publish_debug_topics:=true _use_mag:=false
rostopic echo /imu/rpy/filtered
```

Output:

```yaml
header: 
  seq: 5440
  stamp: 
    secs: 1360242258
    nsecs:  19164474
  frame_id: "imu"
vector: 
  x: 3.13005686173
  y: 0.0160443984394
  z: -1.38870353583
---
```

You're right about `/imu/rpy/raw` not being published by the way. It only gets filled when `use_mag` is `true`. I'm never using the debug topics anyway; instead, the quaternion orientation from `/imu/data` should be used.

closed 2020-01-13 by mintar