CCNYRoboticsLab/imu_tools #99: Compiling madgwick fails

ArashJavan, 2019-12-08:
unfortunately i get following error during build process:

My Env:
ROS: Melodic
OS: Ubuntu 18.04

```
Errors     << imu_filter_madgwick:make /home/pc-milux/catkin_ws/catkin_miomap/logs/imu_filter_madgwick/build.make.001.log                               
/home/pc-milux/catkin_ws/catkin_miomap/src/imu_tools/imu_filter_madgwick/src/imu_filter_nodelet.cpp:39:24: error: expected constructor, destructor, or type conversion before ‘(’ token
 PLUGINLIB_DECLARE_CLASS(imu_filter_madgwick, ImuFilterNodelet, ImuFilterNodelet, nodelet::Nodelet);
```
any idea what is going wrong?

Thanks

closed 2019-12-08 by ArashJavan

numb0824, 2021-01-22:
hi,i have met the same problem .My Env is "ubuntu18.04+ros-melodic".Please help me what should i do?

mintar, 2021-01-22:
Please check out the `melodic` branch.