CCNYRoboticsLab/imu_tools #102: Stateless orientation is ROS dependent

AlexisTM, 2020-01-13:
The StatelessOrientation is dependent on ROS while the rest of the filter is not.

The dependence is only for the types (Vector3 and Quaternion) but no "convenience" function is used.

Should we make the StatelessOrientation ROS independent?

mintar, 2020-01-13:
Actually, some convenience functions are used, for example here:

https://github.com/ccny-ros-pkg/imu_tools/blob/11f405e89925d75b9a17c64d92a139a10ae56118/imu_filter_madgwick/src/stateless_orientation.cpp#L143

In my opinion, the stateless orientation is only useful for one thing: It's used to estimate the initial orientation of the IMU and initializing the filter with it. Before it was added, the filter was initialized with a neutral orientation (0, 0, 0) and took around 5-10 seconds to converge to the actual orientation. So having the stateless orientation for that purpose is useful. For any other purposes I'm not so sure, because setting the parameter "stateless" to true basically disables the whole rest of the filter.

Do you have a use case for using the stateless orientation outside of ROS? If not, I wouldn't spend any effort on making it ROS independent.

AlexisTM, 2020-01-14:
Indeed, I missed the part of using TF2 matrixes & conversions. It is much less trivial to convert it than expected.

We are planning on using the imu_tools in a non-ROS environment where we use it as a shared library right in our position filtering to reduce delays and unpredictability of ROS(1) communications. 

For such use-case, when not initializing the filter with the stateless orientation, the gravity removal fails (is unreliable) until convergence. In the meantime, we will still have the ROS API available and I will convert it & share it if time allows.

closed 2020-01-14 by AlexisTM

mintar, 2020-01-14:
Sure, if you submit a PR, I'm open to accepting it. I just wanted to make sure you really have a use case for this and don't spend your time just on "making it nice" for aesthetic purposes. :)

AlexisTM, 2020-01-14:
But aesthetic is key!  :innocent: (And obviously I have no time for it :eyes:)