CCNYRoboticsLab/imu_tools #106: ROS2 Madgwick

tgreier, 2020-05-13:
Hello,
I have spent some time migrating the melodic branch of the madgwick filter to ROS2. Overall, the filter seems to be functioning. However, when configured for stateful operation (which is how it should be configured), the filtered heading is very unstable under any changes in acceleration (frequent changes of sign in yaw). When configured for stateless, the 'filtered' heading is very stable.  So it seems that the stateless orientation calculation is fine, but something not right in the AHRS filter. Building on x86 Ubuntu (eloquent). I was wondering if you would be able to help diagnose the problem. I am happy to share what I have (although I'm not sure how to do that. create a branch and pull request?). I'm using a memsense 3030 IMU, but I could rosbag the imu output for replication. I've verified the timestamps from the IMU ros node as well as the calculated dt in the madgwick node. Also tried gains between 0.05 and 1.0, all exhibiting same behavior.
Thank you for your help,
Tom

mintar, 2020-05-18:
Hello Tom, 

thanks for porting this to ROS2, seems like a valuable contribution! Some ideas for debugging:

* Does it work with your data on Melodic, but not on your ROS2 fork?
* There's some example data that's known to work in the [imu_filter_madgwick/sample](https://github.com/ccny-ros-pkg/imu_tools/tree/indigo/imu_filter_madgwick/sample) directory. Does that work on your ROS2 fork?

If stuff works on melodic, but not on your fork, there's a bug in your code somewhere. If your data also doesn't work on melodic, there's a problem with your data.

> I am happy to share what I have (although I'm not sure how to do that. create a branch and pull request?)

You could just fork the repository and push your changes to the fork. No need to create a pull request yet.

> I'm using a memsense 3030 IMU, but I could rosbag the imu output for replication

Yes, that would be very helpful.

tgreier, 2020-05-22:
Well, it turns out that units and orientation are extremely important.... :)
Our imu node units and orientation were completely incorrect. I followed posted notes on testing the orientation and units, and then led me to the issues. The filter is now working beautifully.
I have only so far ported the madgwick filter to ROS2. I have not yet ported the Google tests, nor the rviz plugin or complementary filter. Would anyone be interested in the ROS2 madgwick? How best for me to contributed what I've done? Thank you for your help.

mintar, 2020-05-26:
> How best for me to contributed what I've done?

Open a [pull request](https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/about-pull-requests) ("PR").

I've created a new branch "eloquent" in this repository (assuming "eloquent" is the ROS2 version that you're using). Please target your PR against this branch.

mintar, 2020-09-25:
Fixed by #110.

closed 2020-09-25 by mintar