CCNYRoboticsLab/imu_tools #107: imu_tools get wrong for xsens IMU

crankler, 2020-05-23:
Hi, I get 2 wrong display when I use imu_tools,  and I've change the topic as  "/imu/data".
1. when I just launch the IMU node and rviz, I try to compare the display of rviz_plugin_tutorials and rviz_imu_plugin, then I move and you can see the difference as follow:
[image]

2. Then I standing the IMU in the ground and input the command:
> roslaunch imu_complementary_filter complementary_filter.launch

and the red arrow get random beat as follow:
[image]

So what can I do to get rid of this error?
Thank yuor for your reply!

mintar, 2020-05-26:
What am I looking at here? I don't understand exactly what the problem is.

mintar, 2020-09-25:
Closing due to inactivity.

closed 2020-09-25 by mintar