CCNYRoboticsLab/imu_tools #111: roslaunch error

nyangshawbin, 2020-06-01:
Hi, 
I have an `/imu` topic in a rosbag that I would like to use `imu_complementary_filter` to get orientation quaternion. However, I ran into the following 2 errors after playing the rosbag and running the command `roslaunch imu_complementary_filter complementary_filter.launch`:

> ERROR: cannot launch node of type [imu_complementary_filter/complementary_filter_node]: Cannot locate node of type [complementary_filter_node] in package [imu_complementary_filter]. Make sure file exists in package path and permission is set to executable (chmod +x)

> [ERROR] [1590996199.172990714]: Failed to load nodelet [/PhidgetsImuNodelet] of type [phidgets_imu/PhidgetsImuNodelet] even after refreshing the cache: According to the loaded plugin descriptions the class phidgets_imu/PhidgetsImuNodelet with base class type nodelet::Nodelet does not exist. Declared types are  SlamGMappingNodelet depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register depthimage_to_laserscan/DepthImageToLaserScanNodelet hdl_graph_slam/FloorDetectionNodelet hdl_graph_slam/HdlGraphSlamNodelet hdl_graph_slam/PrefilteringNodelet hdl_graph_slam/ScanMatchingOdometryNodelet image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image imu_filter_madgwick/ImuFilterNodelet laser_proc/LaserProcNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid stereo_image_proc/disparity stereo_image_proc/point_cloud2
[ERROR] [1590996199.173059635]: The error before refreshing the cache was: According to the loaded plugin descriptions the class phidgets_imu/PhidgetsImuNodelet with base class type nodelet::Nodelet does not exist. Declared types are  SlamGMappingNodelet depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/register depthimage_to_laserscan/DepthImageToLaserScanNodelet hdl_graph_slam/FloorDetectionNodelet hdl_graph_slam/HdlGraphSlamNodelet hdl_graph_slam/PrefilteringNodelet hdl_graph_slam/ScanMatchingOdometryNodelet image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image imu_filter_madgwick/ImuFilterNodelet laser_proc/LaserProcNodelet nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid stereo_image_proc/disparity stereo_image_proc/point_cloud2
[FATAL] [1590996199.173228558]: Failed to load nodelet '/PhidgetsImuNodelet` of type `phidgets_imu/PhidgetsImuNodelet` to manager `imu_manager'
[PhidgetsImuNodelet-2] process has died [pid 5208, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load phidgets_imu/PhidgetsImuNodelet imu_manager __name:=PhidgetsImuNodelet __log:=/home/shaw/.ros/log/52a9e96e-a3d5-11ea-b7a5-9cb6d0156d0b/PhidgetsImuNodelet-2.log].
log file: /home/shaw/.ros/log/52a9e96e-a3d5-11ea-b7a5-9cb6d0156d0b/PhidgetsImuNodelet-2*.log

Regarding the first error, i have tried `sudo chmod +x complementary_filter_node.cpp` but it doesnt seem to solve the issue. Would appreciate if someone can give me advice on how to solve these issues. Thanks in advance!

mintar, 2020-06-01:
If you want to build from source, you need to build your workspace using `catkin_make` and source the `devel/setup.bash` file. But it's probably easier for you to just install the binaries instead. Then you do not need to clone this repository:

```bash
sudo apt install ros-kinetic-imu-tools   # if you're on kinetic
sudo apt install ros-melodic-imu-tools   # if you're on melodic
sudo apt install ros-noetic-imu-tools    # if you're on noetic
```

nyangshawbin, 2020-06-01:
Yup, building from source works, thanks! But If i would like to use the repo instead, how do i fix the error of: 

> ERROR: cannot launch node of type [imu_complementary_filter/complementary_filter_node]: Cannot locate node of type [complementary_filter_node] in package [imu_complementary_filter]. Make sure file exists in package path and permission is set to executable (chmod +x)

mintar, 2020-06-15:
As I said in my previous comment, you did not build or source your catkin workspace correctly. This is not a problem specific to this package. Please try the [ROS tutorials](http://wiki.ros.org/ROS/Tutorials) first, then ask your question on ROS Answers. See:

https://www.ros.org/support/

closed 2020-06-15 by mintar

numb0824, 2021-01-20:
hi，i have met the same problem，can you tell me what should i do ？

mintar, 2021-01-20:
First, make sure you're not on the `indigo` branch, but on the proper branch for your ROS distro.