CCNYRoboticsLab/imu_tools #112: orientation always returns to same value when using /imu/mag

karsjen, 2020-06-03:
when only using gyroscope data, both filters work fine. However, hen i add magnetometer data however the filter will react to a change in orientation but always returns to its initial state. I am using the MPU-9250 9dof module. This is a sample what my imu/mag data looks like.

header: 
  seq: 5359
  stamp: 
    secs: 1591187353
    nsecs: 391313076
  frame_id: "imu"
magnetic_field: 
  x: -9.9e-06
  y: 5.76e-05
  z: -2.64e-05
magnetic_field_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 5360
  stamp: 
    secs: 1591187353
    nsecs: 400687932
  frame_id: "imu"
magnetic_field: 
  x: -8.85e-06
  y: 5.745e-05
  z: -2.685e-05
magnetic_field_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

mintar, 2020-06-15:
Could you record a short 10s rosbag of all the filter input topics where this problem occurs, and also post your launch file with the parameters that you are using, so I can reproduce the problem?

mintar, 2020-09-25:
Closing due to inactivity.

closed 2020-09-25 by mintar