CCNYRoboticsLab/imu_tools #114: cannot determine gravity direction!

DeamonYang, 2020-06-13:
Hello, I get this WARNING when I use this filter to my IMU data:

The IMU seems to be in free fall or close to magnetic north pole, cannot determine gravity direction!

My input data tipic is  /imu/data_raw and like this:
`---
header: 
  seq: 33192
  stamp: 
    secs: 1592037016
    nsecs: 334757585
  frame_id: "imu_link"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00070735090645
  y: 0.000202100379393
  z: -0.000398882111767
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: -0.347298975847
  y: 0.170656846434
  z: 9.80990094074
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
`
How can I address this problem
TKS for your attention! 
Best wishes.

mintar, 2020-06-14:
That is strange. Could you record a short 10s rosbag where this problem occurs, and also post your launch file with the parameters that you are using, so I can reproduce the problem?

mintar, 2020-09-25:
Closing due to inactivity.

closed 2020-09-25 by mintar