CCNYRoboticsLab/imu_tools #117: Issue with rviz_imu_plugin not loading

moshahin1, 2020-07-23:
Hi, I have added an IMU sensor to my robot (taken from http://gazebosim.org/tutorials?tut=ros_gzplugins). I am getting readings from the IMU but every time I open Rviz to use the `rviz_imu_plugin` to get a visual display, the message below appears: 
```

**Imu**
The class required for this display, 'rviz_imu_plugin/Imu', could not be loaded.
**Error:**
Could not find library corresponding to plugin rviz_imu_plugin/Imu. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
```

I have followed the instructions on the github but as this package contains non-catkin packages I have had to run catkin_make_isolate (could this be the problem?). I am using Ubuntu 18.04 and ROS Melodic. I am new to ROS so I would like to thank you in advance if this issue is on my behalf. Thank you for your help!

mintar, 2020-07-23:
This package does not contain non-catkin packages. Are you sure that you used the `melodic` branch?

mintar, 2020-09-25:
Closing due to inactivity.

closed 2020-09-25 by mintar