CCNYRoboticsLab/imu_tools #123: what is initial output orientation without magnetometer

elgarbe, 2020-09-02:
Hi, I'm testing imu_filter_madgwick with just accelerometer and gyroscope in order to get orientation (2D robot). But I'm not sure what is initial orientation quaternion that the filter output.

mintar, 2020-09-03:
The horizontal orientation is aligned with gravity. Without a magnetometer, there is no way to determine the yaw angle, so it is initialized arbitrarily. However, this was never a problem for my use cases (2D indoor robot navigation + localization). I always run without magnetometer, since it's basically useless when you run it next to powerful electric motors while moving through an environment that's full of ferrous metals. You should configure your localization package (`robot_localization` or `robot_pose_ekf`) to just use the *difference* in yaw orientation, not the absolute value.

elgarbe, 2020-09-03:
ok, I understand now. thank!
BTW may I ask for a RL configuration using just imu_filter and wheel encoder odometry? I'm not sure how to configure imu0 and odom0, what part of each message sensor use, same case as yours, indoor navigation and localization.

mintar, 2020-09-04:
Yep, I have a robot_localization example here:

https://github.com/dfki-ric/mir_robot/blob/19627a4d57dfc769cd45fb17314bb063ebd9418d/mir_gazebo/config/ekf.yaml

https://github.com/dfki-ric/mir_robot/blob/melodic/mir_gazebo/launch/includes/ekf.launch.xml

You can try it in Gazebo, just follow the instructions here: https://github.com/dfki-ric/mir_robot

closed 2020-09-04 by mintar