CCNYRoboticsLab/imu_tools #124: Support rviz plugins for ros2.

tgreier, 2020-09-28:
Request to add rviz imu plugin for ros2 as well as a plugin for Magnetic Field plugin.
In our testing with ros2 using the madgwick filter, it is very useful to have a vector representation of the magnetic field in order to compare with the orientation being calculated by the filter. There were some distortions in the orientation resulting from the motors in our system. By visualizing the magnetic field, we could determine that the heading distortions were a result of magnetic field distortion, and not from a problem within the filter.

https://github.com/ccny-ros-pkg/imu_tools/blob/9145900554fe5079bca80e9b73a847247268c4e8/rviz_imu_plugin/CMakeLists.txt#L1

mintar, 2020-09-28:
Pull request welcome!

mintar, 2020-10-03:
Closed via #124. Thank you!

closed 2020-10-03 by mintar