CCNYRoboticsLab/imu_tools #128: Imcompatability with ROS Noetic

zarathustr, 2020-10-30:
The build on the new ROS release Neotic seems failed simply because of the deprecated macro PLUGINLIB_DECLARE_CLASS which belongs to the PluginLib.  The following issue #https://github.com/KumarRobotics/msckf_vio/pull/72#issue-241485636 mentions this problem. A new merge should be perform by replacing PLUGINLIB_DECLARE_CLASS with PLUGINLIB_EXPORT_CLASS, which is located in multiple source files related to nodelet.

mintar, 2020-10-30:
You should use the branch `noetic`, everything is working there. You also don't need to build from source, you can just type `sudo apt install ros-noetic-imu-tools`.

zarathustr, 2020-10-30:
> You should use the branch `noetic`, everything is working there. You also don't need to build from source, you can just type `sudo apt install ros-noetic-imu-tools`.

Thanks for your message. But I am using the source build ROS for Mac with newest version of neotic therefore only this means is available.

mintar, 2020-10-30:
What I mean is this:

```bash
roscd imu_tools
git checkout noetic
```

In the `noetic` branch of this repo, PLUGINLIB_DECLARE_CLASS has already been replaced by PLUGINLIB_EXPORT_CLASS three years ago. You are using an outdated branch.

zarathustr, 2020-10-30:
> What I mean is this:
> 
> ```shell
> roscd imu_tools
> git checkout noetic
> ```
> 
> In the `noetic` branch of this repo, PLUGINLIB_DECLARE_CLASS has already been replaced by PLUGINLIB_EXPORT_CLASS three years ago. You are using an outdated branch.

OK mintar, thanks very much!

mintar, 2020-10-30:
Glad to help!

closed 2020-10-30 by mintar