CCNYRoboticsLab/imu_tools #133: Bad compilation results using Clang

zarathustr, 2020-12-26:
Hi Admin,

It seems that the compatability with the Clang compiler is not satisfactory. I tested the compilation from ROS-Indigo/OS X 10.9.5/Clang-5.1.0 to ROS-Noetic/OS X 10.15/Clang-11.0. The results show that for early releases, the linking is good but for later releases the linker always fails. The compilers even faill to find common libraries e.g. message_filter, rosconsole, etc. This may be because of the syntax option of the codes? Do you have any suggestions about this issue? Many thanks!

zarathustr, 2021-01-16:
Hi @mintar , I also tested the imu_tools on ROS-Noetic/OS X 10.14.5/Clang-11.0 and encountered the same errors. 

The details are during linking process:

Errors     << imu_complementary_filter:make /Users/zarathustra/catkin_ws/logs/imu_complementary_filter/build.make.002.log            
In file included from /Users/zarathustra/catkin_ws/src/imu_tools/imu_complementary_filter/src/complementary_filter_ros.cpp:32:
In file included from /Users/zarathustra/catkin_ws/src/imu_tools/imu_complementary_filter/include/imu_complementary_filter/complementary_filter_ros.h:37:
In file included from /Users/zarathustra/ros_noetic/install_isolated/include/message_filters/subscriber.h:38:
In file included from /Users/zarathustra/ros_noetic/install_isolated/include/ros/ros.h:45:
In file included from /Users/zarathustra/ros_noetic/install_isolated/include/ros/node_handle.h:32:
In file included from /Users/zarathustra/ros_noetic/install_isolated/include/ros/publisher.h:35:
/usr/local/include/boost/bind.hpp:36:1: warning: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior. [-W#pragma-messages]
BOOST_PRAGMA_MESSAGE(
^
/usr/local/include/boost/config/pragma_message.hpp:24:34: note: expanded from macro 'BOOST_PRAGMA_MESSAGE'
# define BOOST_PRAGMA_MESSAGE(x) _Pragma(BOOST_STRINGIZE(message(x)))
                                 ^
<scratch space>:414:2: note: expanded from here
 message("The practice of declaring the Bind placeholders (_1, _2, ...) " "in the global namespace is deprecated. Please use " "<boost/bind/bind.hpp> + using namespace boost::placeholders, " "or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.")
 ^
1 warning generated.
Undefined symbols for architecture x86_64:
  "message_filters::Connection::disconnect()", referenced from:
      void message_filters::Synchronizer<message_filters::sync_policies::ApproximateTime<sensor_msgs::Imu_<std::__1::allocator<void> >, sensor_msgs::MagneticField_<std::__1::allocator<void> >, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType> >::connectInput<message_filters::Subscriber<sensor_msgs::Imu_<std::__1::allocator<void> > >, message_filters::Subscriber<sensor_msgs::MagneticField_<std::__1::allocator<void> > >, message_filters::NullFilter<message_filters::NullType>, message_filters::NullFilter<message_filters::NullType>, message_filters::NullFilter<message_filters::NullType>, message_filters::NullFilter<message_filters::NullType>, message_filters::NullFilter<message_filters::NullType>, message_filters::NullFilter<message_filters::NullType>, message_filters::NullFilter<message_filters::NullType> >(message_filters::Subscriber<sensor_msgs::Imu_<std::__1::allocator<void> > >&, message_filters::Subscriber<sensor_msgs::MagneticField_<std::__1::allocator<void> > >&, message_filters::NullFilter<message_filters::NullType>&, message_filters::NullFilter<message_filters::NullType>&, message_filters::NullFilter<message_filters::NullType>&, message_filters::NullFilter<message_filters::NullType>&, message_filters::NullFilter<message_filters::NullType>&, message_filters::NullFilter<message_filters::NullType>&, message_filters::NullFilter<message_filters::NullType>&) in complementary_filter_ros.cpp.o
      message_filters::Synchronizer<message_filters::sync_policies::ApproximateTime<sensor_msgs::Imu_<std::__1::allocator<void> >, sensor_msgs::MagneticField_<std::__1::allocator<void> >, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType> >::~Synchronizer() in complementary_filter_ros.cpp.o
  "message_filters::Connection::Connection(boost::function<void ()> const&)", referenced from:
      message_filters::Connection message_filters::SimpleFilter<sensor_msgs::Imu_<std::__1::allocator<void> > >::registerCallback<imu_tools::ComplementaryFilterROS, boost::shared_ptr<sensor_msgs::Imu_<std::__1::allocator<void> > const> const&>(void (imu_tools::ComplementaryFilterROS::*)(boost::shared_ptr<sensor_msgs::Imu_<std::__1::allocator<void> > const> const&), imu_tools::ComplementaryFilterROS*) in complementary_filter_ros.cpp.o
      message_filters::Connection message_filters::SimpleFilter<sensor_msgs::Imu_<std::__1::allocator<void> > >::registerCallback<ros::MessageEvent<sensor_msgs::Imu_<std::__1::allocator<void> > const> const&>(boost::function<void (ros::MessageEvent<sensor_msgs::Imu_<std::__1::allocator<void> > const> const&)> const&) in complementary_filter_ros.cpp.o
      message_filters::Connection message_filters::SimpleFilter<sensor_msgs::MagneticField_<std::__1::allocator<void> > >::registerCallback<ros::MessageEvent<sensor_msgs::MagneticField_<std::__1::allocator<void> > const> const&>(boost::function<void (ros::MessageEvent<sensor_msgs::MagneticField_<std::__1::allocator<void> > const> const&)> const&) in complementary_filter_ros.cpp.o
      message_filters::Connection message_filters::Signal9<sensor_msgs::Imu_<std::__1::allocator<void> >, sensor_msgs::MagneticField_<std::__1::allocator<void> >, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_fil [... cut ...]
      void ros::Publisher::publish<std_msgs::Bool_<std::__1::allocator<void> > >(std_msgs::Bool_<std::__1::allocator<void> > const&) const in complementary_filter_ros.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/zarathustra/catkin_ws/devel/.private/imu_complementary_filter/lib/libcomplementary_filter.dylib] Error 1
make[1]: *** [CMakeFiles/complementary_filter.dir/all] Error 2
make: *** [all] Error 2

zarathustr, 2021-01-16:
@mintar I also managed to resolve this issue, which are listed below:

In the CMakeList.txt, it seems that the linking mechanism is incorrect for Clang compiler, but works exactly for GCC. My revision is:

## Declare a cpp library
add_library(complementary_filter
  src/complementary_filter.cpp
  src/complementary_filter_ros.cpp
  src/complementary_filter_node.cpp
  include/imu_complementary_filter/complementary_filter.h
  include/imu_complementary_filter/complementary_filter_ros.h
)

# create complementary_filter_node executable
add_executable(complementary_filter_node
  src/complementary_filter_node.cpp
  src/complementary_filter.cpp
  src/complementary_filter_ros.cpp
)
target_link_libraries(complementary_filter_node ${catkin_LIBRARIES})
target_link_libraries(complementary_filter ${catkin_LIBRARIES})

mintar, 2021-01-16:
That does not seem correct. With your change, the `complementary_filter_node` is not linked to the `complementary_filter` library at all. There has to be a way to use dynamic linking, even with clang on OS X, right?

Also, why did you add `src/complementary_filter_node.cpp` to the library? That file only contains the `main` function. The library doesn't need it.

zarathustr, 2021-01-16:
Yes there should be a way for dynamic library. If I do not add that file, there will be a strange error shown below:

`Undefined symbols for architecture x86_64:`
`"imu_tools::ComplementaryFilterROS::ComplementaryFilterROS(ros::NodeHandle const&, ros::NodeHandle const&)", referenced from:`
`      _main in complementary_filter_node.cpp.o`
`  "imu_tools::ComplementaryFilterROS::~ComplementaryFilterROS()", referenced from:`
`      _main in complementary_filter_node.cpp.o`
`ld: symbol(s) not found for architecture x86_64`
`clang: error: linker command failed with exit code 1 (use -v to see invocation)`
`make[2]: ***` `[/Users/zarathustra/catkin_ws/devel/.private/imu_complementary_filter/lib/imu_complementary_filter/complementary_filter_node] Error 1`
`make[1]: *** [CMakeFiles/complementary_filter_node.dir/all] Error 2`
`make: *** [all] Error 2`
`

mintar, 2021-01-16:
Yes, these symbols are defined in the dynamic library. Somehow the linker doesn't find the library.

By the way, you can create code blocks in comments with **three back-ticks**: https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code

zarathustr, 2021-01-16:
@mintar Thanks for your kind comments!

closed 2021-11-10 by mintar