CCNYRoboticsLab/imu_tools #134: [ WARN] [1609424955.602349746]: Still waiting for data on topics /imu/data_raw and /imu/mag...

gugiek, 2020-12-31:
Hi,

I am trying to get the madgwick filter working on a raspberry pi 4 with ubuntu mate 20.04 installed.
For testing purposes, I am using two dummy publisher nodes that publish a constant message to the /imu/data_raw and /imu/mag topics.

When activating the madgwick filter node I keep getting the warning message that is written in the title and, if I listen to the /imu/data topic, there is nothing being published.

Do you have any idea what could be the problem here? 

I have attached the messages format and result from running rostopic and rosnode list.

[image]
[image]
[image]
[image]

Thanks !

3ricS, 2021-02-14:
Hi,
I ran into the same problem. Somehow the topic `/imu/mag` is expected to have the message type `geometry_msgs/Vector3Stamped` when using ROS Noetic with Ubuntu 20.04. Normally the message type of this topic should be `sensor_msgs/MagneticField`.

I solved this problem by explicitly setting the parameter `use_magnetic_field_msg` to true. Pass this after the rosrun command: `_use_magnetic_field_msg:=true`.
This tells the Node to expect messages of the type `sensor_msgs/MagneticField` at the topic `/imu/mag`.

I hope this also solves your problem.

gugiek, 2021-02-14:
Thanks a lot for the answer!
I ended up offloading imu filtering to a microcontroller, but I'll definitively look into this solution...   

Cheers!

closed 2021-02-14 by gugiek

mintar, 2021-05-05:
You seem to have resolved this problem, but I'll just mention this here in case somebody stumbles upon this issue in the future.

There seems to be some confusion here. To clear it up:

* The parameter `use_magnetic_field_msg` has been removed for all ROS distros >= Lunar in 2017 (f5be6024). If you're using the binary install, you should be good. If you're using the source install, make sure you are on the correct git branch. I suspect @3ricS was using the `indigo` branch or something like that.
* You should always use `sensor_msgs/MagneticField` message, not `geometry_msgs/Vector3Stamped`. It looks like @gugiek was trying to publish a `Vector3Stamped`, but `imu_filter_madgwick` expects a `MagneticField` message.