CCNYRoboticsLab/imu_tools #135: WARNING: The CMakeLists.txt of the metapackage 'imu_tools' contains non standard content. Use the content of the following file instead: /home/lyd/imu_catkin_ws/build/catkin_generated/metapackages/imu_tools/CMakeLists.txt

nuaalyd, 2021-01-19:
You should remove the folder "imu_tools" before catkin_make, otherwise an error will occur like this:

`CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build  a non-homogeneous workspace without isolation.  Try the  'catkin_make_isolated' command instead.`

And I wonder if anyone has such a problem.

mintar, 2021-01-19:
You're probably on the "indigo" branch, which is *really* old. Switch to the "kinetic" branch instead.

numb0824, 2021-01-22:
i have met the same problem,what should i do to solve this problem

numb0824, 2021-01-22:
and my branch is melodic

mintar, 2021-01-22:
Ok, I've fixed it in 32fac84b. Please pull the newest version of the `melodic` branch to fix this error.

commit 3396cfeb (2021-01-22): Fix "non standard content" warning in imu_tools metapackage

On catkin_make, the cmake_policy line produced the following error:

    WARNING: The CMakeLists.txt of the metapackage 'imu_tools' contains non standard content.

Compilation with catkin_make_isolated or catkin (from catkin-tools)
worked though.

Just updating the minimum CMake version to 3.5.1 should also be enough
to fix the buildfarm warning that we fixed in 01058cf5, since 3.5.1
doesn't require setting this cmake policy any more. Therefore, we also
remove it from the other (non-metapackage) packages.

Fixes #135.
closed 2021-01-22 by mintar