CCNYRoboticsLab/imu_tools #140: linear acceleration of zero breaks the filter

timonegk, 2021-05-29:
I am using the imu complementary filter and noticed that when the linear acceleration input to the filter is [0, 0, 0], the filter returns [nan, nan, nan, nan] as orientation and continues to do so forever.
The input of [0, 0, 0] happens in my case in simulation when the robot falls. I think the best way to handle this would be to simple skip a linear acceleration of [0, 0, 0].

robertogl, 2021-11-09:
That is because of the normalization of a zero norm vector. Sure, skipping that case sounds a reasonable solution.

robertogl, 2021-11-10:
Not sure if we want to track the orientation of a falling object, but we can do it (within a reasonable amount of time) just by using the prediction. So, if acceleration has zero norm then we do not use the correction. Note that a perfect zero norm would only happen in simulation, in real life we would have a small norm (<9.81) which I think is already handled by the adaptive gain.

TuanLBT, 2024-06-09:
@robertogl Hi, can you recommend me the way to skip this case. Im new to Ros and got the same issue that the filter give NaN value (0, 0, 0) leads the rviz to shutdown sometime.