CCNYRoboticsLab/imu_tools #151: Moving this repo to ros-perception?

mintar, 2021-11-10:
Dear @robertogl , what do you think about moving this repo to the official [ros-perception](https://github.com/ros-perception) GitHub group? We would still be maintainers, all repo and git URLs etc. would continue to function (because GitHub will add a redirect), the issues, forks, stars etc. would also be transferred. The advantages would be:

1. Better visibility by making the repo part of the official ros-perception group from OSRF.
2. Ensuring that OSRF can give permissions to a new maintainer in case the current maintainers go AWOL.

Also a nice side benefit would be that I could finally get proper permissions to do things like *finally* update the default branch of the repo. It's still set to `indigo`, which went EOL in 2014. This causes regular bugs when people try to compile from source and don't realize that they are using an outdated version of the repo...

robertogl, 2021-11-10:
It sounds like a good idea to me. Are there any disadvantages at all?

mintar, 2021-11-11:
No, there are no disadvantages. The prerequisite is that you need to have "owner" permissions in the `ccny-ros-pkg` organization or "admin" permissions to the `imu_tools` repository. Could you check?

The actual transfer will happen like this: https://docs.github.com/en/repositories/creating-and-managing-repositories/transferring-a-repository#transferring-a-repository-owned-by-your-organization

In the meanwhile, I've created a new post on discourse: https://discourse.ros.org/t/moving-imu-tools-to-ros-perception-github-organization/23067 . Let's hope somebody with the proper permissions to `ros-perception` answers it.

robertogl, 2021-11-11:
Unfortunately I do not have nay of those. I can ask to get 'admin' role to this repo. I will let you know.

mintar, 2021-11-11:
Yes, it would be really great if you could do that!

mintar, 2021-11-19:
One of the `ros-perception` maintainers, @clalancette, has responded in the discourse thread. We could move this repository as soon as you get the proper permissions. It's something I've been trying to do [since 2013](https://groups.google.com/g/ros-sig-drivers/c/sHWShPO3s2o/m/FahPxVB8lAQJ), and I think we should go through with it now.

@robertogl : Could you try again getting the proper permissions?

@ehoxha91 : You've mentioned in #94 that you know who might have the proper permissions to this repo. Could you try again asking for the permissions?

@idryanov : I know you haven't been active here for a couple of years, but if by any chance you read this, could you give me or @robertogl the permissions?

I would hate to fork this repository into ros-perception instead of doing a proper move. We would lose all the github stars, issues and the network of forks, but I guess it's something we have to do at some point if we can't get the permission issue sorted out.

ehoxha91, 2021-11-19:
@mintar I have the password of the account, but the issue is that whenever I’m not in the office I cannot login (2-Factor Authentication and I don’t have the access to the email address to get the code). For some reason I don’t need 2FA when I am in the Lab. As soon as I go in there I will give @robertogl and you the admin privileges.

mintar, 2021-11-19:
@ehoxha91 : Great, please keep us updated!

> For some reason I don’t need 2FA when I am in the Lab.

That's probably because you (or someone else) already logged in from that PC using 2FA. Let's just hope nobody clears the browser cache or reinstalls the machine, otherwise you could lose access to this GitHub organization permanently...

I just hope you don't need 2FA to give @robertogl and me the admin privileges...

ehoxha91, 2021-11-19:
@mintar the thing is that the machine I use on lab is a new one, just recently used. This issue started few months ago. If I need 2FA to give the admin privileges then we will need access to one of these emails:

"We just sent your authentication code via email to c**************@gmail.com and r**************@gmail.com."

mintar, 2021-12-02:
@ehoxha91 : Any updates?

ehoxha91, 2021-12-02:
@mintar Prof. Xiao is coming to the lab today so I’ll try to do it, and ask him for access to the email too.

mintar, 2021-12-02:
Perfect, thanks!

ehoxha91, 2021-12-03:
@mintar please contact me at:

ehoxha@ccny.cuny.edu

mintar, 2021-12-21:
Prof. Xiao (@ccny-ros-pkg) has agreed to move this repo to ros-perception. He will either give me admin access or transfer the repository to @clalancette.

@clalancette: If you get an invitation email from GitHub that asks whether you want to allow transferring this repo to your user account, could you accept and then move it to ros-perception? I think this detour is necessary because the repo cannot be transferred directly to ros-perception by users without repository creation permissions on ros-perception.

The invitation email expires after 1 day, that's why I'm giving you an advance warning. :)

clalancette, 2021-12-21:
Yes, I can do that.  But do note that I'm away from December 24 - Jan 3, so I may or may not see it during that time.

mintar, 2021-12-21:
Understandable, I'm on vacation during that time as well. This might be the worst time of the year to do this... let's see!

mintar, 2022-04-07:
After many, many discussions and iterations, we've decided not to move this repo to `ros-perception`, but to the newly created `CCNYRoboticsLab` organization, so I'm closing this issue.

closed 2022-04-07 by mintar