CCNYRoboticsLab/imu_tools #152: Madgwick filter: Still waiting for data on topics /imu/data_raw and /imu/mag

jcremona, 2021-11-10:
I am trying to run `imu_filter_madgwick` to fuse imu and mag. Unfortunately, the filter never starts:
`[ WARN] [1636574811.749864237]: Still waiting for data on topics /imu/data_raw and /imu/mag`

```
PARAMETERS
 * /imu_with_mag/publish_tf: False
 * /imu_with_mag/use_mag: True
 * /imu_with_mag/use_magnetic_field_msg: True
 * /imu_with_mag/world_frame: enu
 * /rosdistro: noetic
 * /rosversion: 1.15.13
```
This is the [rosbag](https://github.com/ccny-ros-pkg/imu_tools/files/7515672/_2021-11-10-15-10-42.zip) I'm trying to run. Topic remapping shouldn't be a problem. IMHO it is a synchronization problem, but I am not sure how sync works. Has anyone had a similar problem?

mintar, 2021-11-11:
There are two problems:

* The IMU data should be published on the topic `/imu/data_raw`, not `/imu`
* The time stamps of the imu and mag topic have to roughly match. Currently, it looks like they are 12 seconds apart:

```
$ rostopic echo imu -b _2021-11-10-15-10-42.bag -n 1
header: 
  seq: 1
  stamp: 
    secs: 1636567843
    nsecs:  67188978
  frame_id: "imu_link"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 1.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.049162328243255615
  y: -1.1736339330673218
  z: 0.4956202208995819
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 8.106766821289062
  y: 1.4526818572998046
  z: 1.5927426544189451
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
$ rostopic echo imu/mag -b _2021-11-10-15-10-42.bag -n 1
no field named mag in /mag
header: 
  seq: 1
  stamp: 
    secs: 1636567831
    nsecs: 482614994
  frame_id: "imu_link"
magnetic_field: 
  x: -1.675877571105957e-05
  y: -3.852567911148071e-06
  z: 1.2643646240234374e-05
magnetic_field_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
```

jcremona, 2021-11-11:
Thanks @mintar . I didn't notice that, because timestamps in rosbag are fine. 
[image]
But timestamps in headers are several seconds apart.

closed 2021-11-11 by jcremona