CCNYRoboticsLab/imu_tools #154: Release for Foxy & Galactic

shrijitsingh99, 2022-01-10:
Any plans to release this to Foxy and Galactic? Currently its only available on Eloquent.

mintar, 2022-01-10:
Yes, I'll do the releases soon!

CollinNHays, 2022-01-19:
Figured it's worth noting that the eloquent branch will compile on galactic. About to try it out!

sskorol, 2022-01-24:
@mintar tried it on foxy. I can successfully build packages and see rviz2 plugins. However, when I try launching `imu_complementary_filter`, it fails with the following error:
```shell
❯ ros2 launch imu_complementary_filter complementary_filter.launch.py
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2022-01-24-11-24-15-875419-User-13299
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [complementary_filter_node-1]: process started with pid [13301]
[complementary_filter_node-1] /home/user/ros2_workspaces/imu_tools/install/imu_complementary_filter/lib/imu_complementary_filter/complementary_filter_node: symbol lookup error: /home/user/ros2_workspaces/imu_tools/install/imu_complementary_filter/lib/imu_complementary_filter/complementary_filter_node: undefined symbol: _ZN7tf2_ros20TransformBroadcaster13sendTransformERKN13geometry_msgs3msg17TransformStamped_ISaIvEEE
[ERROR] [complementary_filter_node-1]: process has died [pid 13301, exit code 127, cmd '/home/user/ros2_workspaces/imu_tools/install/imu_complementary_filter/lib/imu_complementary_filter/complementary_filter_node --ros-args -r __node:=complementary_filter_node --params-file /tmp/launch_params_dyxh5m8u --params-file /tmp/launch_params_v5yajl9g --params-file /tmp/launch_params_2edbexvq --params-file /tmp/launch_params_jm128cvj --params-file /tmp/launch_params_7uivoekl --params-file /tmp/launch_params_xoxm20ax'].
```

Launch file:
```python
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.substitutions import LaunchConfiguration

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='imu_complementary_filter',
            executable='complementary_filter_node',
            name='complementary_filter_node',
            output='screen',
            parameters=[
                {'fixed_frame': 'base_link'},
                {'do_bias_estimation': True},
                {'do_adaptive_gain': True},
                {'use_mag': True},
                {'gain_acc': 0.01},
                {'gain_mag': 0.01},
                {"publish_tf": True}
            ])
        ])
```

If I comment `tf_broadcaster_.sendTransform(transform);` in `complementary_filter_ros.cpp`, launch fails with a bit different error:
```shell
❯ ros2 launch imu_complementary_filter complementary_filter.launch.py
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2022-01-24-11-30-40-655003-User-15099
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [complementary_filter_node-1]: process started with pid [15101]
[complementary_filter_node-1] /home/user/ros2_workspaces/imu_tools/install/imu_complementary_filter/lib/imu_complementary_filter/complementary_filter_node: symbol lookup error: /home/user/ros2_workspaces/imu_tools/install/imu_complementary_filter/lib/imu_complementary_filter/complementary_filter_node: undefined symbol: _ZN15message_filters10ConnectionC1ERKSt8functionIFvvEE
[ERROR] [complementary_filter_node-1]: process has died [pid 15101, exit code 127, cmd '/home/user/ros2_workspaces/imu_tools/install/imu_complementary_filter/lib/imu_complementary_filter/complementary_filter_node --ros-args -r __node:=complementary_filter_node --params-file /tmp/launch_params_fost97hj --params-file /tmp/launch_params_v2dypj8k --params-file /tmp/launch_params_tnfn_ypa --params-file /tmp/launch_params_dk_a9r98 --params-file /tmp/launch_params_o3tq21mv --params-file /tmp/launch_params_2dtawt0n --params-file /tmp/launch_params_i9y4867i'].
```

Any thoughts on how to fix it?

SammyRamone, 2022-02-02:
Please also release for Rolling :slightly_smiling_face: 
I just tested the Eloquent branch and it works on Ubuntu 20.04 with Rolling

HusamZain, 2022-03-07:
@sskorol  can you show me how did you make the rviz plug in work. sorry I'm new to this 
I'm trying just to visualize imu data in rviz.
  I cloned the repo and tried to build it, but I got an error
is there any changes that i should make before building the package ?
thank you in advance

HappySamuel, 2022-03-17:
Hi

Will there be any debian release? then everyone can install via (`sudo apt install ros-foxy-imu-tools`)

Best,
Samuel

mintar, 2022-03-17:
Yes, I'm currently still waiting for this repo to be transferred to a different GitHub organization. After that, I'll make a ROS2 release.

sskorol, 2022-03-17:
> @sskorol  can you show me how did you make the rviz plug in work. sorry I'm new to this 
> 
> I'm trying just to visualize imu data in rviz.
> 
>   I cloned the repo and tried to build it, but I got an error
> 
> is there any changes that i should make before building the package ?
> 
> thank you in advance

Sorry, I can't give you details atm. You know what happens in Ukraine. I'm far away from my PC.

mintar, 2022-04-19:
Released into foxy + galactic, rolling coming soon:

* https://github.com/ros/rosdistro/pull/32797
* https://github.com/ros/rosdistro/pull/32798

closed 2022-04-19 by mintar