CCNYRoboticsLab/imu_tools #156: use_magnetic_field_msg does not work

TYXDG, 2022-04-27:
Hello! I'm doing some experiments with jackal, and I want to fuse /imu/ data_raw and /imu/mag to get the absolute orientation of the vehicle. But since the message type given by Jackal for /imu/mag is geometry_MSgs /Vector3Stamped, I set use_magnetic_field_msg to false, But the "Client [/imu_filter] wants topic/imu/mag to have datatype/md5sum [sensor_msgs/MagneticField/2f3b0b43eed0c9501de0fa3ff89a45aa], but our version has [geometry_msgs/Vector3Stamped/7b324c7325e683bf02a9b14b01090ec7]" is still prompted. Can someone give me some help? thank you

mintar, 2022-04-27:
The parameter `use_magnetic_field_msg` has been removed for all ROS distros >= Lunar in 2017 (https://github.com/CCNYRoboticsLab/imu_tools/commit/f5be6024415d240c7d2b9b28d540f82d058fc41a). It was only there to enable backwards compatibility to pre-2017 code.

Ideally, you would update your robot to publish `sensor_msgs/MagneticField` messages. If that's not possible, you could use the `mag_convert.py` script from #116 to re-publish the Vector3Stamped messages as MagneticField and feed that topic into `imu_filter_madgwick`.

It's ironic that you're having problems with the `sensor_msgs/MagneticField` on a Jackal robot, because Jackal is made by Clearpath Robotics, and it was @mikepurvis from Clearpath who initiated the switch of this repo to MagneticField in #43 7 years ago...

TYXDG, 2022-04-27:
Thanks for your advice!

closed 2022-04-27 by TYXDG