CCNYRoboticsLab/imu_tools #158: How to use on ROS2 Dashing?

saitenntaisei, 2022-05-04:
I want to use madgwick filter on ROS2 Dashing. 
However, I cannot find  a branch for Dashing.
I know that ROS2 Dashing is already out of date but I cannot change ROS2 version currently.
Therefore is there anyway to use madgwick filter on ROS2 Dashing?
Is it possible to use madgewick filter for other version of ROS2 distribution?

P.S. 
It is my first time to post issue on open git, thus there may be some mistakes. 
I would like to know them if I have. Thanks.

mintar, 2022-05-04:
There are no released binaries of this repo for Dashing, and there never will be, so you have to build from source. You could try checking out the `eloquent` git branch and try if it compiles from source on ROS2 Dashing. With some luck, it should work.

saitenntaisei, 2022-06-17:
It does not work... maybe I'm lacking luck.  But Thanks.

closed 2022-06-17 by saitenntaisei