CCNYRoboticsLab/imu_tools #160: Bad results from filter after time stamp jump

nachumlerner, 2022-05-15:
Hi
Our robot doesn't have a permanent internet connection so when it is turned on its clock starts from. 1.1.2016. After startup our robot starts calculating orientation by using madgwick filter. After a few minutes we get wifi signal and the clock adjusts to correct time. When the clock adjusts we get a big jump in orientation From the filter even if the robot is stationary on the floor.
If we restart the filter we will get the correct orientation (can be seen in image in /sensors/imu_filtered1).
Any advice?
Thanks
[image]

mintar, 2022-05-16:
>  Any advice?

Just avoid these clock jumps. :)

Sorry to not be more helpful, but you're feeding false information into the filter. The IMU data that goes into the filter says that there is a time stamp of 6 years in the sensor data, and the filter does the best it can based on that information - garbage in, garbage out. It doesn't seem that this should be handled in the madgwick filter, but in the data that goes into it.

nachumlerner, 2022-05-16:
Is there a way to initialize the yaw angle after relaunch so it will be continuous with the angle before the jump?

mintar, 2022-05-16:
There's the parameter `yaw_offset` that you could probably abuse for this purpose. However, I'm afraid you'll get plenty of other errors in other ROS nodes that are caused by the time jump, such as TF buffers being invalidated and so on. Can't you add a Real-Time Clock to your robot so that you can fix the root of all problems, instead of adding tons of workarounds all over the place?

nachumlerner, 2022-05-23:
Unfortunately we can't add anything to the robot, thanks for the advice

mintar, 2022-05-24:
That's a bummer. I dug a bit deeper, and it turns out that you can fix your problem with `imu_filter_madgwick` by setting the parameter `constant_dt` to the rate at which your IMU publishes. If that parameter is set, `imu_filter_madgwick` will no longer look at the time stamp of the messages, but uses the given $\Delta t$ in its calculations.

closed 2022-08-19 by mintar