CCNYRoboticsLab/imu_tools #177: Problem with magnetic use complementary filter

tanphanzero, 2023-04-15:
I use complementary filter for IMU and use Rviz to simulate the orientation by data from publish_tf. It's very OK but when I set use_mag to true, my base_link frame always skew North. Is it a normal thing? How can I fix that? Thank you

mintar, 2023-04-18:
You should probably visualize the magnetic field message that your sensor outputs. I guess either the IMU doesn't publish a proper magnetic reading at all, or there are some magnetic fields that distort the sensor reading.

Personally, I have never had any luck with using a magnetometer on an indoor robot - you're better off switching it off. The earth magnetic field is very weak, and disturbances from electric motors on the robot, steel beams in the building etc. are much too strong.

If you're working on an outdoor robot and you can mount the magnetometer far away from any motors (e.g., on a car or boat), it's worth looking into.

mintar, 2023-11-14:
Closing due to lack of activity.

closed 2023-11-14 by mintar