CCNYRoboticsLab/imu_tools #184: output of imu_filter_madgwick is skewed over time

phu251099, 2023-08-26:
I am currently using a filter to filter out the raw IMU value from the D435i camera but it seems that the yaw value on the continuous Z increases (or decreases) over time even though the camera mounted on the robot is stationary. Can anyone help me with this problem? Thank you guys in advance.

https://github.com/CCNYRoboticsLab/imu_tools/assets/133739987/805d41d4-d056-4983-b444-7e544742db89

mintar, 2023-08-28:
There's probably a drift in the IMU yaw angle of the D435i camera. You could plot the IMU output to find out.

Kaiser-Wang, 2023-09-28:
I had the same issue. The D435i camera only outputs angular velocity and acceleration. It doesn't have a magnetometer and therefore the use_mag is set to false while using imu_filter_madgwick to get orientation.

The angular velocity is not perfectly 0 rad/s and contains small noises (within 0.01 rad/s) when standing still. The acceleration is also not perfectly 9.81 m/s^2 when sitting on a flat table (it's about 9.7 m/s^2). Both angular velocity and acceleration are steady (don't drift over time) but the calculated orientation (especially yaw around world vertical axis) drifts over time (about a few degrees every minute).

The inaccurate acceleration can lead to errors in the estimation of roll and pitch angles but shouldn't cause the yaw angle to drift over time. The angular velocity contains some noises that can accumulate errors in orientation over time but shouldn't these noises be filtered by the imu_filter_madgwick package?