CCNYRoboticsLab/imu_tools #189: Incorrect variance calculation

r0gi, 2023-11-24:
This package includes a reconfigurable param `orientation_stddev` that gets [squared](https://github.com/CCNYRoboticsLab/imu_tools/blob/77aaa91247fbc8ee08e9d9431490a138e52821d1/imu_filter_madgwick/src/imu_filter_ros.cpp#L438-L439) to become a variance value, which is published as part of the `orientation_covariance` field of the [Imu](https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Imu.html) message. 

However, I don't think that the orientation covariance should be treated in the linear sense, since orientation wraps around each axis. There doesn't seem to be a lot of clarity or resources online as to how it should be used in ROS, and https://www.ros.org/reps/rep-0103.html doesn't provide clarity either, but it makes most sense that the variance for each axis should be based on a [wrapped normal distribution](https://en.wikipedia.org/wiki/Wrapped_normal_distribution). In such a distribution, the _variance_ should range from 0.0-1.0, and the variance is also not simply the square of the standard distribution.

r0gi, 2023-11-24:
I found REP 145: Conventions for IMU Sensor Drivers
https://www.ros.org/reps/rep-0145.html:
>~orientation_stddev (double)
Square root of the orientation_covariance diagonal elements in rad. Overrides any values reported by the sensor.

So this package matches the REP, but I think the REP is wrong. Either way, I'm not sure how practical it is to change this since the value here is manually set anyway.

mintar, 2023-12-20:
To be honest, the representation of the IMU noise as a covariance matrix is not good anyway. Any serious application that cares about the IMU noise will probably use the Allan variance model - look at https://github.com/ori-drs/allan_variance_ros and its references for a good start into this rabbit hole.

closed 2024-02-06 by mintar