CCNYRoboticsLab/imu_tools #194: "odom" passed to lookupTransform argument target_frame does not exist.

sahilsaqi, 2024-04-01:
I am using Ros Noetic on Ubuntu 20.04 LTS

and I installed this library for integration of IMU (``mpu6050_driver``)  with Hector SLAM.
basically as you know mpu6050_driver provides ``/imu/raw/data`` so  I need to filter this using madgwick filter, 

so here is  the steps I am taking 

1. ``roscore``
2. ``roslaunch rplidar_ros rplidar_a1.launch``
3. ``roslaunch hector_slam_launch tutorial.launch``
4. ``rosrun imu_filter_madgwick imu_filter_node _use_mag:=false``

also my rostopics are 
``aiot@ubuntu:~/ros_slam$ rostopic list
/ImuFilter/parameter_descriptions
/ImuFilter/parameter_updates
/clicked_point
/imu/data
/imu/data_raw
/imu/orientation_filtered
/imu/rpy/filtered
/imu/rpy/raw
/initialpose
/map
/map_metadata
/map_updates
/move_base_simple/goal
/poseupdate
/rosout
/rosout_agg
/scan
/slam_cloud
/slam_out_pose
/syscommand
/tf
/tf_static
/trajectory``

output of  /imu/data:
``aiot@ubuntu:~/ros_slam$ rostopic echo /imu/data
header:
  seq: 1474
  stamp:
    secs: 1711980942
    nsecs: 312890118
  frame_id: "imu"
orientation:
  x: 0.006631039159660429
  y: -0.0016873730512080361
  z: 0.3952459307362719
  w: 0.9185498551130837
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
  x: -0.0015987749211490154
  y: 0.003197549842298031
  z: -0.0014655436389148235
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration:
  x: -0.16645386815071106
  y: 0.19040405750274658
  z: 9.604028701782227
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header:
  seq: 1475
  stamp:
    secs: 1711980942
    nsecs: 393032546
  frame_id: "imu"
orientation:
[... 58 lines cut ...]
[ WARN] [1711980225.369746148]: Lookup would require extrapolation 0.080009458s into the future.  Requested time 1711980225.072674274 but the latest data is at time 1711980224.992664814, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980225.472171343]: Lookup would require extrapolation 0.120006251s into the future.  Requested time 1711980225.192680597 but the latest data is at time 1711980225.072674274, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980225.573861582]: Lookup would require extrapolation 0.080132030s into the future.  Requested time 1711980225.272812605 but the latest data is at time 1711980225.192680597, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980225.674720716]: Lookup would require extrapolation 0.119895437s into the future.  Requested time 1711980225.392708063 but the latest data is at time 1711980225.272812605, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980225.777249095]: Lookup would require extrapolation 0.119946715s into the future.  Requested time 1711980225.512654781 but the latest data is at time 1711980225.392708063, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980225.880725189]: Lookup would require extrapolation 0.084809806s into the future.  Requested time 1711980225.597464561 but the latest data is at time 1711980225.512654781, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980225.982833461]: Lookup would require extrapolation 0.115185718s into the future.  Requested time 1711980225.712650299 but the latest data is at time 1711980225.597464561, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980226.085786577]: Lookup would require extrapolation 0.080025068s into the future.  Requested time 1711980225.792675257 but the latest data is at time 1711980225.712650299, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980226.190565345]: Lookup would require extrapolation 0.120348452s into the future.  Requested time 1711980225.913023710 but the latest data is at time 1711980225.792675257, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980226.291782957]: Lookup would require extrapolation 0.079824700s into the future.  Requested time 1711980225.992848396 but the latest data is at time 1711980225.913023710, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980226.393573861]: Lookup would require extrapolation 0.119806752s into the future.  Requested time 1711980226.112655163 but the latest data is at time 1711980225.992848396, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980226.495224210]: Lookup would require extrapolation 0.120330507s into the future.  Requested time 1711980226.232985735 but the latest data is at time 1711980226.112655163, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980226.596839652]: Lookup would require extrapolation 0.080337381s into the future.  Requested time 1711980226.313323021 but the latest data is at time 1711980226.232985735, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980226.705611534]: Lookup would require extrapolation 0.120179341s into the future.  Requested time 1711980226.433502436 but the latest data is at time 1711980226.313323021, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980226.808164413]: Lookup would require extrapolation 0.079160094s into the future.  Requested time 1711980226.512662411 but the latest data is at time 1711980226.433502436, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980226.910210981]: Lookup would require extrapolation 0.120081639s into the future.  Requested time 1711980226.632744074 but the latest data is at time 1711980226.512662411, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980227.012716990]: Lookup would require extrapolation 0.080063957s into the future.  Requested time 1711980226.712808132 but the latest data is at time 1711980226.632744074, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980227.115880603]: Lookup would require extrapolation 0.120569282s into the future.  Requested time 1711980226.833377361 but the latest data is at time 1711980226.712808132, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980227.217370787]: Lookup would require extrapolation 0.080151308s into the future.  Requested time 1711980226.913528681 but the latest data is at time 1711980226.833377361, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980227.323792392]: Lookup would require extrapolation 0.119351089s into the future.  Requested time 1711980227.032879829 but the latest data is at time 1711980226.913528681, when looking up transform from frame [imu] to frame [odom]
[ WARN] [1711980227.426363214]: Lookup would require extrapolation 0.119881011s into the future. ```

if you need any clarification please reply soon,

i need help here

mintar, 2024-04-02:
The imu_filter_madgwick shouldn't publish the `odom -> imu` transform. This is just for visualization/debugging. You can disable it like this:

```
rosrun imu_filter_madgwick imu_filter_node _use_mag:=false _publish_tf:=false
```

The `odom -> imu` transform published by the IMU only contains the rotation, not the translation. On a wheeled robot, you need to use `robot_localization` or something equivalent to fuse the IMU values and the wheel odometry to obtain a TF transform like this:

```mermaid
  graph LR;
      IMU-driver -->|/imu/data_raw| imu_filter_madgwick;
      imu_filter_madgwick -->|/imu/data| robot_localization;
      your-robot-driver -->|/odom| robot_localization;
      robot_localization --> |tf-transform| hector_mapping;
```

... where the `/odom` topic has the message type [nav_msgs/Odometry](http://docs.ros.org/api/nav_msgs/html/msg/Odometry.html), `/imu/data_raw` and `/imu/data` topics have type [sensor_msgs/Imu](http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html), and the "tf-transform" is a TF transform like `odom -> base_footprint` that is required (along with other TF transforms from the URDF) by hector_mapping.

sahilsaqi, 2024-04-03:
> The imu_filter_madgwick shouldn't publish the `odom -> imu` transform. This is just for visualization/debugging. You can disable it like this:
> 
> ```
> rosrun imu_filter_madgwick imu_filter_node _use_mag:=false _publish_tf:=false
> ```
> 
> The `odom -> imu` transform published by the IMU only contains the rotation, not the translation. On a wheeled robot, you need to use `robot_localization` or something equivalent to fuse the IMU values and the wheel odometry to obtain a TF transform like this:
> 
> ```mermaid
>   graph TD;
>       IMU-driver -->|/imu/data_raw| imu_filter_madgwick;
>       imu_filter_madgwick -->|/imu/data| robot_localization;
>       your-robot-driver -->|/odom| robot_localization;
>       robot_localization --> |tf-transform| hector_mapping;
> ```
> ... where the `/odom` topic has the message type [nav_msgs/Odometry](http://docs.ros.org/api/nav_msgs/html/msg/Odometry.html), `/imu/data_raw` and `/imu/data` topics have type [sensor_msgs/Imu](http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html), and the "tf-transform" is a TF transform like `odom -> base_footprint` that is required (along with other TF transforms from the URDF) by hector_mapping.

Actually I am using handheld device, so in my case should I still need to use robot localization as I need to operate the device on hand or helmet.

and also is there any problem like 
while i ``echo /imu/data -> frame
-id = "imu",
same  frame is using /imu/data_raw``

sahilsaqi, 2024-04-03:
> The imu_filter_madgwick shouldn't publish the `odom -> imu` transform. This is just for visualization/debugging. You can disable it like this:
>
> ```
> rosrun imu_filter_madgwick imu_filter_node [...]

okay I just tried this approach 

you can see the graph below, but there is still some problem.

[image]

i have used the ``static transform publisher`` instead of ``robot localization`` because i am using hand held device so i don't have any wheeled robot 

here i am providing the static publisher launch file. 

``<launch>

<node pkg="tf2_ros" type="static_transform_publisher" name="test_static_pub_imu" args="0 0 0 0 0 0 1 base_link imu" />

  
  <rosparam param="test_tf2/tf_valid"
            file="$(find test_tf2)/test/test_tf_valid.yaml" />
  <node pkg="tf2_ros" type="static_transform_publisher"
        name="test_static_pub_param_valid"
        args="test_tf2/tf_valid" />

  
  <rosparam param="test_tf2/tf_invalid"
            file="$(find test_tf2)/test/test_tf_invalid.yaml" />
  <node pkg="tf2_ros" type="static_transform_publisher"
        name="test_static_pub_param_invalid"
        args="test_tf2/tf_invalid" />

  
  <node pkg="tf2_ros" type="static_transform_publisher"
        name="test_static_pub_param_null"
        args="test_tf2/tf_null" />

  <test test-name="test_static_publisher" pkg="test_tf2"
        type="test_static_publisher" args="--text" />

  <test test-name="test_static_publisher_py"
        pkg="test_tf2"
        type="test_static_publisher.py"
        launch-prefix="python$(env ROS_PYTHON_VERSION)"/>
</launch>``

i am pretty sure that there is something wrong.

that is why I am getting drift on map while moving a bit or rotate the handheld device

[image]

these are my topics 

``aiot@ubuntu:~$ rostopic list
/ImuFilter/parameter_descriptions
/ImuFilter/parameter_updates
/clicked_point
/imu/data
/imu/data_raw
/initialpose
/map
/map_metadata
/map_updates
/move_base_simple/goal
/poseupdate
/rosout
/rosout_agg
/scan
/slam_cloud
/slam_out_pose
/syscommand
/tf
/tf_static
/trajectory``

mintar, 2024-04-03:
So if you look at the [hector_mapping documentation](http://wiki.ros.org/hector_mapping), you'll see the following:

### Required tf Transforms

`<the frame attached to incoming scans> → base_frame` - usually a fixed value, broadcast periodically by a [robot_state_publisher](http://wiki.ros.org/robot_state_publisher), or a tf [static_transform_publisher](http://wiki.ros.org/tf#static_transform_publisher).

... and:

`~pub_map_odom_transform` (bool, default: true) -- Determine if the map->odom transform should be published by the system. 

So it looks like in your case, you'll need the `base_link` -> `laser` transform, which you can publish via a `static_transform_publisher`. Also, if you set the `~pub_map_odom_transform` parameter (or just leave it as default), `hector_mapping` will publish map to odom, so you mustn't do that yourself.

Also, I don't see any option in `hector_slam` to use IMU input, so you probably don't need this package here at all.

Anyway, this discussion has nothing to do with `imu_tools`, so please ask any follow-up questions on https://robotics.stackexchange.com/.

Good luck!

closed 2026-09-17 by mintar