CCNYRoboticsLab/imu_tools #201: Yaw estimation with madgwick filter has a bias.

faustodidomenico, 2024-07-03:
Hello, I'm trying to estimate the orientation using Madgwick filter in a simulated rosbag created with Gazebo, but the estimation has some errors compared to the ground-truth. I compare the results using [https://github.com/MichaelGrupp/evo](url). I am testing the filter accuracy. This rosbag consists of only a few seconds with the robot still. It does not move. Yaw estimation has a clear bias of approximately 1.5° and pitch has some errors too. 

[image]

I would like to know the reason for what is generating the bias in the yaw. I ran the filter with declination parameter equals to -0.150072136 (in radians) corresponding to -8.5985° configured in gazebo magnetometer plugin.

The IMU is simulated with the Intel RealSense Gazebo ROS plugin [https://github.com/pal-robotics/realsense_gazebo_plugin](url) and the magnetometer with Hector Gazebo Ros Magnetic plugin [https://github.com/tu-darmstadt-ros-pkg/hector_gazebo](url)
The sensors configuration:
```
<sensor name="${camera_name}imu" type="imu">
        <always_on>true</always_on>
        <update_rate>${gyro_fps}</update_rate>
        <topic>${camera_name}/imu/sample</topic>
        <plugin name="${camera_name}imu" filename="libgazebo_ros_imu_sensor.so">
          <topicName>${camera_name}/imu/sample</topicName>
          <bodyName>${camera_name}_link</bodyName>
          <frameName>${gyro_optical_frame}</frameName>
          <updateRateHZ>${gyro_fps}</updateRateHZ>
          <gaussianNoise>0.0</gaussianNoise>
          <xyzOffset>0 0 0</xyzOffset>
          <rpyOffset>0 0 0</rpyOffset>
          <!-- Nota(Dante): añadido ver http://gazebosim.org/tutorials?tut=ros_gzplugins#IMU(GazeboRosImu) -->
          <initialOrientationAsReference>false</initialOrientationAsReference>
        </plugin>
      </sensor>
    </xacro:if>
    <!-- Accel and gyro -->
    <xacro:unless value="${unite_imu_method == 'copy' or unite_imu_method == 'linear_interpolation'}">
      <sensor name="${camera_name}accel" type="imu">
        <always_on>true</always_on>
        <update_rate>${accel_fps}</update_rate>
        <topic>${camera_name}/accel/sample</topic>
        <plugin name="${topics_ns}_accel_plugin" filename="librealsense_gazebo_accel_sensor.so">
          <topicName>${camera_name}/accel/sample</topicName>
          <bodyName>${camera_name}_link</bodyName>
          <updateRateHZ>${accel_fps}</updateRateHZ>
          <gaussianNoise>0.0</gaussianNoise>
          <xyzOffset>0 0 0</xyzOffset>
          <rpyOffset>0 0 0</rpyOffset>
          <frameName>${accel_optical_frame}</frameName>
          <initialOrientationAsReference>false</initialOrientationAsReference>
        </plugin>
        <pose>0 0 0 0 0 0</pose>
      </sensor>
      <sensor name="${camera_name}gyro" type="imu">
        <always_on>true</always_on>
        <update_rate>${gyro_fps}</update_rate>
        <topic>${camera_name}/gyro/sample</topic>
        <plugin name="${topics_ns}_gyro_plugin" filename="librealsense_gazebo_gyro_sensor.so">
          <topicName>${camera_name}/gyro/sample</topicName>
          <bodyName>${camera_name}_link</bodyName>
          <updateRateHZ>${gyro_fps}</updateRateHZ>
          <gaussianNoise>0.0</gaussianNoise>
          <xyzOffset>0 0 0</xyzOffset>
          <rpyOffset>0 0 0</rpyOffset>
          <frameName>${gyro_optical_frame}</frameName>
          <initialOrientationAsReference>false</initialOrientationAsReference>
        </plugin>
```
```
<plugin name="hector_magnetometer" filename="libhector_gazebo_ros_magnetic.so">
      <bodyName>magnetometer_link</bodyName>
      <frameId>magnetometer_link</frameId>
      <topicName>magnetometer/fix</topicName>
      <updateRate>300.0</updateRate>
      <referenceHeading>90</referenceHeading>
      <declination>-8.5985</declination>
      <inclination>-37.8919</inclination>
      <magnitude>22.5782e-06</magnitude>
      <gaussianNoise>0 0 0</gaussianNoise>
      <useMagneticFieldMsgs>true</useMagneticFieldMsgs>
      <useWorldSphericalCoordinates>false</useWorldSphericalCoordinates>
    </plugin>
```

This issue is related to one I created a few months ago #191

mintar, 2024-07-04:
There's a clear bias in the input data. Here's the linear acceleration. `linear_acceleration.z` is around 9.8, which is to be expected because of gravity, but x is  around -3 instead of 0. I.e., the robot is tilted forward, and pitch should be negative, which is the case; in other words, your ground truth is wrong, and the filter is right.

[image]

Similarly, as you wrote, you have added a bias to the magnetometer in the form of declination and inclination, so your magnetometer data also has a strong bias, which leads to the bias in yaw reported by the filter.

I would say the filter is working as expected: It gives the best estimate of the orientation based on your input data; the ground truth doesn't reflect the input data. Your ground truth is incorrect. I hope this helps.

faustodidomenico, 2024-07-04:
Hello, thank you for your response. I'm sorry, I forgot to mention in the issue that the IMU and the ground truth are in different coordinates systems. So, I make a rotation to the filter results to compare them. The rotation is -17.189° in pitch angle:
[image]

If I compare the results without doing this rotation, yaw and roll remain the same, but pitch is 17.189° higher. You can see this image and the other one I used in the issue:
[image]

Also, I forgot to mention that if I use the default value for declination parameter (equals to 0), I still have a bias, but greater (approximately 7°). 

I mention all of this, because maybe with this information you come to another conclusion.

Thank you again, for your time.

closed 2024-07-29 by faustodidomenico