CCNYRoboticsLab/imu_tools #212: How to transform RPY values from imu_sensor frame to base_link frame?

aatefi2, 2025-01-03:
Hi,

I am utilizing `imu_tools/imu_filter_madgwick/src/imu_filter_ros.cpp` (ros2 iron) to obtain roll, pitch, and yaw (RPY) from the angular velocity and linear acceleration of an IMU sensor (which lacks a magnetometer). I would like to transform RPY from imu_sensor frame to base_link frame. I have modified `fixed_frame_` from "odom" to "base_link" and also `declare_parameter("reverse_tf", true, descriptor)`. How can I adjust the code so that imu/data (or imu/rpy/filtered) is published in the base_link frame with `header.frame_id` set to base_link?

Thanks,
Abbas

mintar, 2025-01-09:
The TF publishing functionality of this repo is very basic and was only intended for debugging. I'm afraid this package cannot do directly what you want it to do.

I assume there is a fixed transform between `base_link` and your `imu_frame`? In that case, it doesn't make sense to set the `fixed_frame`  to `base_link`, because the rotation from `base_link` to `imu_frame` is static.

It only makes sense to set the `fixed_frame` to something outside the robot (like `map` or `odom`), and the `imu_frame` must be the frame in which the IMU measurements are reported. This package doesn't look at any TF transformations at all, but instead it just leaves the translation as (0, 0, 0) and publishes the rotation as TF. Since there usually is a translation between the robot and the fixed frame, this is wrong and only useful for debugging.

What most people do instead is feed the IMU message (which measures rotation) together with another sensor reading like wheel odometry (for measuring translation) into a package such as `robot_localization` which then calculates the proper TF transform from `odom` to `base_link`.

aatefi2, 2025-01-09:
Thank you for your reply. The IMU sensor can pivot up and down relative to the base_link frame. I applied your code to four ROSBAG files, and it accurately calculated the roll and pitch values in the IMU frame. I modified your code (see below) to publish roll and pitch in the base_link frame, and it successfully published the values (imu/rpy/filtered) for all ROSBAG files. However, I did receive one warning message at the beginning of one of the ROSBAG files: `[WARN] [1736442165.545740346] [imu_filter]: Could not transform RPY to base_link: "base_link" passed to lookupTransform argument target_frame does not exist. `

#Code
```
  if (publish_debug_topics_)
    {
        // Step 1: Extract RPY in the imu_msg_raw frame
        geometry_msgs::msg::Vector3Stamped rpy_imu;
        tf2::Matrix3x3(tf2::Quaternion(q1, q2, q3, q0)).getRPY(rpy_imu.vector.x, rpy_imu.vector.y, rpy_imu.vector.z);
        
        // Step 2: Retrieve the transformation from imu_sensor frame to base_link frame
        try
        {
            geometry_msgs::msg::TransformStamped transform;

            transform = tf_buffer_ -> lookupTransform("base_link", imu_msg_raw->header.frame_id, tf2::TimePointZero, tf2::durationFromSec(0.1));

            // Step 3: Convert the transformation to a tf2 quaternion
            tf2::Quaternion imu_to_base_quat;
            tf2::fromMsg(transform.transform.rotation, imu_to_base_quat);

            // Step 4: Convert RPY (imu_msg_raw frame) to a quaternion
            tf2::Quaternion rpy_quat;
            rpy_quat.setRPY(rpy_imu.vector.x, rpy_imu.vector.y, rpy_imu.vector.z);

            // Step 5: Transform the RPY quaternion to the base_link frame
            tf2::Quaternion rpy_transformed_quat = imu_to_base_quat * rpy_quat.inverse();

            // Step 6: Extract the transformed RPY
            double roll, pitch, yaw;
            tf2::Matrix3x3(rpy_transformed_quat).getRPY(roll, pitch, yaw);

            // Step 7: Publish the transformed RPY
            geometry_msgs::msg::Vector3Stamped rpy_base_link;
            rpy_base_link.vector.x = roll * (180.0 / M_PI);;//deg
            rpy_base_link.vector.y = pitch * (180.0 / M_PI);;//deg
            rpy_base_link.vector.z = yaw * (180.0 / M_PI);;//deg
            rpy_base_link.header.stamp = imu_msg_raw->header.stamp;
            rpy_base_link.header.frame_id = "base_link";
            rpy_filtered_debug_publisher_->publish(rpy_base_link);            
        }
```

I am wondering did I correctly modify your code or not? or that warning message can cause any issue?

mintar, 2025-01-10:
Looks good! Regarding the warning: Yes, this can be a problem. You should call `waitForTransform` before `lookupTransform` and only proceed if the transform was found within the timeout.