CCNYRoboticsLab/imu_tools #214: remapping /imu/data_raw topic

hassan-attar, 2025-04-30:
Hi,

I am trying to run the `imu_filter_madgwick_node` using this command:

```bash
ros2 run imu_filter_madgwick imu_filter_madgwick_node --ros-args -r /imu/data_raw:=camera/raw
```

here is the output:
```plaintext
[INFO] [1746045756.253783756] [imu_filter_madgwick]: Starting ImuFilter
[INFO] [1746045756.254445603] [imu_filter_madgwick]: Using dt computed from message headers
[INFO] [1746045756.254493317] [imu_filter_madgwick]: The gravity vector is kept in the IMU message.
[INFO] [1746045756.254608265] [imu_filter_madgwick]: Imu filter gain set to 0.100000
[INFO] [1746045756.254653546] [imu_filter_madgwick]: Gyro drift bias set to 0.000000
[INFO] [1746045756.254679147] [imu_filter_madgwick]: Magnetometer bias values: 0.000000 0.000000 0.000000
[WARN] [1746045766.260863376] [imu_filter_madgwick]: Still waiting for data on topics imu/data_raw and imu/mag...
^C[INFO] [1746045767.528567143] [rclcpp]: signal_handler(signum=2)
```

but I am not able to remap the `/imu/data_raw` topic.
Can you please guide me on that?

rituraj-navindgikar, 2025-06-23:
`ros2 run imu_filter_madgwick imu_filter_madgwick_node \
  --ros-args \
    -p use_sim_time:=true \
    -p use_mag:=false \
    -p gain:=0.1 \
    -p zeta:=0.0 \
    -p world_frame:=enu \
    -p base_link_frame:=base_link \
    -p imu_frame:=imu_link \
    -p output_rate:=200.0 \
    -p queue_size:=50 \
    -r imu/data_raw:=/imu/imu_aligned \
    -r imu/data:=/imu/filtered
`

I am pasting this as it is. This worked for me 
Hope it helps!

usef-wkb, 2025-07-07:
> ```
>  ros2 run imu_filter_madgwick imu_filter_madgwick_node \
>   --ros-args \
>     -p use_sim_time:=true \
>     -p use_mag:=false \
>     -p gain:=0.1 \
>     -p zeta:=0.0 \
>     -p world_frame:=enu \
>     -p base_link_frame:=base_link \
>     -p imu_frame:=imu_link \
>     -p output_rate:=200.0 \
>     -p queue_size:=50 \
>     -r imu/data_raw:=/imu/fixed \
>     -r imu/data:=/imu/filtered
> ```
> 
> I copy pasted mine as it is. Hope it helps!

This won't work, and remapping is not available as the subscriber for both topics (imu/data_raw and imu/mag) are not rclcpp::subscription ptr. They are cpp smart ptr

Jquinny, 2025-09-29:
> > ```
> >  ros2 run imu_filter_madgwick imu_filter_madgwick_node \
> >   --ros-args \
> >     -p use_sim_time:=true \
> >     -p use_mag:=false \
> >     -p gain:=0.1 \
> >     -p zeta:=0.0 \
> >     -p world_frame:=enu \
> >     -p base_link_frame:=base_link \
> >     -p imu_frame:=imu_link \
> >     -p output_rate:=200.0 \
> >     -p queue_size:=50 \
> >     -r imu/data_raw:=/imu/fixed \
> >     -r imu/data:=/imu/filtered
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I copy pasted mine as it is. Hope it helps!
> 
> This won't work, and remapping is not available as the subscriber for both topics (imu/data_raw and imu/mag) are not rclcpp::subscription ptr. They are message_filters::Subscriber

I am also dealing with this issue, standard remapping is not working. I would like to avoid having to remap my imu data topic name in my publisher node if possible

rituraj-navindgikar, 2025-09-29:
> This won't work, and remapping is not available as the subscriber for both topics (imu/data_raw and imu/mag) are not rclcpp::subscription ptr. They are message_filters::Subscriber

ROS 2 still applies topic remapping before the subscriber (of any kind) is constructed

So even though message_filters::Subscriber isn’t an rclcpp::Subscription, it still looks up the topic name through the remapping table

That’s why my command did work — the remap is applied at the ROS graph level, not at the raw subscriber pointer level.

Jquinny, 2025-09-29:
> ROS 2 still applies topic remapping before the subscriber (of any kind) is constructed
>
> So even though message_filters::Subscriber isn’t an rclcpp::Subscription, it still looks up the topic name [...]

You are absolutely right, did a fresh test with a clean install of the package and ran with remapping and it worked. The warning message with the hardcoded topic name was throwing me off

closed 2026-09-17 by mintar